LowPowerLab/RFM69 #27: Reduce battery power drain by enabling address filtering on both sync and address bytes

fabienroyer, 2015-05-02:
The current implementation of the library doesn't leverage the RFM69's ability to filter variable-length packets based on the sync and address bytes: this severely impacts the battery life of all the nodes within the same network because all the nodes on the network "wake up" whenever a node transmits data.

While filtering on the sync byte alone does enable the 'promiscuous mode' feature of the library, it feels like 'promiscuous mode' is a corner-case scenario (troubleshooting / network monitoring) which should not impact the much more common scenario where the radio should only "wake up" the MCU when necessary.

I believe that address filtering on both the sync and address bytes should be enabled by default in the library as it would eliminate unnecessary battery drain in the most common usage scenario.

LowPowerLab, 2015-05-02:
You are correct that the radio would wake the MCU each time a packet is received from the same network and passes CRC. I'd like to see your claim "severely impacts the battery life" by numbers. Remember you are keeping the radio in RX mode to receive anything at all. So that right there is a 16mA continuous penalty. My guess is that RX continuous mode "severely impacts the battery life" by an order of magnitude more at least. But you have a point that in most cases it makes no sense to wake the MCU.

fabienroyer, 2015-05-02:
Per the RFM69 datasheet, leveraging the "Listen Mode" option of the "Stand By" mode would mitigate the 16mA continuous drain of "Receive Mode", reducing the drain down to 1.25mA (page 12).

4.3. Listen Mode
The circuit can be set to Listen mode, by setting ListenOn in RegOpMode to 1 while in Standby mode. In this mode, RFM69W spends most of the time in Idle mode, during which only the RC oscillator runs. Periodically the receiver is woken up and listens for an RF signal. If a wanted signal is detected, the receiver is kept on and the data is demodulated. Otherwise, if a wanted signal hasn't been detected after a pre-defined period of time, the receiver is disabled until the next time period. This periodical Rx wake-up requirement is very common in low power applications. On RFM69W it is handled locally by the Listen mode block without using uC resources or energy.

LowPowerLab, 2015-05-02:
I appreciate your research. Have you tried the "Listen Mode"?
I have. You can reduce the RX consumption to almost nothing. But you also reduce the chance of receiving to almost nothing. It's all a tradeoff. Also IMHO the effort to implement this mode is not really worth it since you can cycle RX mode in the main MCU code directly.
I suggest these forum discussions of Listen Mode:
https://lowpowerlab.com/forum/index.php/topic,775.0.html
https://lowpowerlab.com/forum/index.php/topic,793.0.html

fabienroyer, 2015-05-02:
Thanks for the forum links. I haven't experimented with "Listen Mode" yet but I understand the trade-offs.

Setting the "Listen Mode" question aside for now, I still believe that address filtering makes sense :)

Thank you.

LowPowerLab, 2015-05-19:
I reviewed this proposition again. To implement it will require a few breaking changes. The return will not be a dramatic improvement although I agree address filtering does make sense. So for the time being I will keep the library the way it is now and will consider address filtering for a future implementation. Thanks for your suggestions.

closed 2015-05-19 by LowPowerLab