LowPowerLab/RFM69 #31: RFM69HW Get lock when sending an ACK

emmaanuel, 2015-06-19:
Hello,

I'm getting an error with the ACK with a radio.setHighPower() and radio.sendACK after receiving a message.

The code is very simple, and my RFM69 work at 868MHz:
```
void setup() {
  radio.initialize(FREQUENCY, NODEID, NETWORKID);
  radio.encrypt(ENCRYPTKEY);
  radio.promiscuous(true);
  radio.setHighPower();
  radio.sleep();
}

void loop() {
if (radio.receiveDone())
  {
    for (byte i = 0; i < radio.DATALEN; i++)
      buff[i] = (char)radio.DATA[i];
    buff[radio.DATALEN] = '\0';
    if ((radio.TARGETID == NODEID) && radio.ACKRequested())
    {
     radio.sendACK();   
    }
...
```
The code get lock on the sendAck line (tested with debug log on each line)
If I comment the radio.sendACK();   every thing work fine.
If a comment the radio.setHighPower();  every thing work fine.

It's look like both are not compatible.

abouillot, 2015-06-21:
Have you tried to put some demay before the sendACK? I'm using a delay(3). This is something I saw in some sample code.

emmaanuel, 2015-06-21:
I've just tried, and I still have a lock on my radio.sendACK();
I'm waiting new piece of RFM69HW, to be sure it's not a problem from this module...

LowPowerLab, 2015-06-21:
The library is extensively tested to work, you might have an issue in your code. setHighPower is only to be used with RFM69HW/HCW modules.
You should start with known working example sketches like the Node and Gateway examples.
Please post any further issues you're having in the lowpowerlab.com/forum

closed 2015-06-21 by LowPowerLab