LowPowerLab/RFM69 #35: will LoRa be supported

einsiedlerkrebs, 2016-03-02:
Hey Felix, 

thanks for your great work on the RFM-libs. We are about to deploy them in our bee-monitoring project @hiveeyes. Since we are trying to have a small energy footprint and still support infrastructural outstanding locations we are very interessted in the Long Range modules from hopeRF. 

Are you planning to write an arduino lib for them? 

Thanks a lot

einsiedlerkrebs

LowPowerLab, 2016-03-02:
There is a variant of RadioHead available for these that is modified to support Moteinos out of box. The  library is linked on the bottom of this page: http://lowpowerlab.com/moteino/#lora
The library contains example sketches and more info is also available at: http://www.airspayce.com/mikem/arduino/RadioHead/classRH__RF95.html
There is an intent to make a library (or wrap RadioHead) for the purpose of making it compatible with the API of RFM69 and hence allow wireless programming via the same familiar interfaces but no effort has been put into this yet.
Either way the LoRa Moteinos are becoming more and more popular.

closed 2016-03-02 by LowPowerLab

einsiedlerkrebs, 2016-03-02:
Okay, thanks for that.