LowPowerLab/RFM69 #36: Corrupted packets when using encrypted transmission

vvvlc, 2016-03-06:
Hi,

I was testing you library and it is very cool, however I was observing strange phenomena.
When you send data from one node to another while they don't have same encryption keys,
sometimes a corrupted packed is received. 
I would expect that no packet should be received because keys are different.

For example: 
I was sending a 61 byte long packet where  first 4 bytes were random numbers and rest were numbers in sequence 4, 5, 6, ..., 59, 60, with encryption key #define KEY         "SAMPLEenCRYPTkEY"
-5 106 6 0 4 5 6 7 8 ... 54 55 56 57 58 59 60

I received this corrupted packet  (encryption key is "sampleEncryptKey")
[10]  [RX_RSSI:-31] [Data len:61] [ERROR in data] 
23 103 148 98 11 223 13 205 200 113 33 153 105 153 140 31 142 100 63 65 101 212 25 9 76 80 169 39 61 251 253 59 54 113 155 172 166 44 21 238 
210 92 245 218 91 85 123 78 185 167 0 56 236 37 102 137 119 254 224 173 205 

I attached two sketches (sender and receiver as example) let it run for some time.
Receiver sketch is displaying only corrupted packets.

I would assume if the encryption keys are not same than packets cannot be received because of invalid crc. 
Does that mean that CRC collision happens more frequently than it is expected?

When you use promiscuous mode it is more visible.
[sketches.zip](https://github.com/LowPowerLab/RFM69/files/160312/sketches.zip)

Btw I'm using Funky V3 instead of Moteino. It is visible in setup() function.
Regards
Vitek

LowPowerLab, 2016-03-08:
This is normal. As long as the CRC matches the packet is decrypted. The radio has no way of knowing a packet was encrypted with a different Key.

closed 2016-03-08 by LowPowerLab

vvvlc, 2016-03-09:
Ok, I made some investigation in datasheet and you are right, thx for quick
assessment.