LowPowerLab/RFM69 #43: ListenMode ISR hook

karrots, 2016-05-25:
It would be nice to attach a function to the ListenMode ISR from user space. As it stands right now the userland code has no way of knowing if a WDT ISR or ListenMode ISR triggered the wake up.

LowPowerLab, 2016-06-06:
Sorry for taking so long to ack this, definitely something I will look into!

LowPowerLab, 2016-06-06:
Any suggestion how you'd like this implemented?

karrots, 2016-06-07:
I guess there are two ways to go about this. 

1. Introduce a function one can call to check if ListenMode initiated the wake-up.
2. Introduce a function to register a callback so the user can do what they will.

LowPowerLab, 2016-06-08:
@rrobinet I can't find anything from kiwi.
@karrots I guess I would prefer a callback, something [like this](http://stackoverflow.com/a/2298865). Although if someone hooks a large function that is not appropriate to call from an ISR, any thoughts about that?

Where would you put the hook to be place in the [listenModeInterruptHandler](https://github.com/LowPowerLab/RFM69/blob/ListenMode/RFM69.cpp#L911)?

I think it would make sense to put this at the end of it. It should then be a minimal function that just sets a flag in user land so they know it woke from ListenMode. Does that sound reasonable?

karrots, 2016-06-08:
Like @rrobinet said either need a way when the uC wakes to check if the radio did the waking. Or a call back like you referenced so the user can set their own variable or some such.

My personal need wouldn't need the call back to be called inside the ISR but it may need to be to keep from getting interrupted and losing state. Like you mention this can get dangerous for those not familiar with doing a minimal amount of items inside the ISR. 

If you go the callback route it should be the very last thing called inside your handler it would seem.

rrobinet, 2016-06-09:
I am not sure we are talking about the same problem, but I refer to a SPI interrupt  issue while using an ethernet shield together with rfm69 transceiver

I am using a  'old' rfm library version (not the ATC one)  patched with the solution proposed by kiwi which solves this interrupt conflict.

Robert

karrots, 2016-06-10:
@rrobinet No conflicts here. Just no way to tell if ListenMode woke from sleep or some other signal on one of the two interrupt pins that can wake the uC from sleep. The RFM69 library already hooks the needed ISR for its own purposes.