LowPowerLab/RFM69 #52: sendFrame is called even if canSend is false?

lpercifield, 2016-10-19:
Just checking to make sure I have the logic right:
The send() function checks to make sure there isn't channel activity with the canSend() function. send() then blocks until the RF69_CSMA_LIMIT_MS is exceeded or if the channel clears. Looking at the while loop, it seems that if the RF69_CSMA_LIMIT_MS is exceeded but, the channel is still active , the sendFrame() proceeds.

I'm suggesting something like the following:
  while (!canSend() && millis() - now < RF69_CSMA_LIMIT_MS) receiveDone();
  if(!canSend() && millis()-now > RF69_CSMA_LIMIT_MS){
    return;
  }else{
      sendFrame(toAddress, buffer, bufferSize, requestACK, false);
  }

Please let me know if I'm wrong or on the wrong track

LowPowerLab, 2016-10-19:
Your observation is correct. However the sendFrame after a timeout is a last resort attempt to send something rather than nothing. Under normal conditions any FSK channel will clear way before the timeout occurs. I would rather keep the current logic, the only risk being introducing a collision to a busy channel, but I consider that a low risk given the channel should not be flooded for more than the timeout, not without allowing gaps or hopping to other channels. If that is actually the case there are other bigger problems elsewhere with whatever is flooding it. If you can identify a use case where this is a big problem let me know and I can reopen this and look at what's the best way to mitigate it. Thanks!

closed 2016-10-19 by LowPowerLab

lpercifield, 2016-10-19:
I'm working on a battery powered application, and it seems, though yet to identify the issue completely, that I have a channel flooding issue related to low battery capacity. So I started digging into everything and noticed the above condition. I'll follow up with results once I get it sorted.
Thanks for the great code, BTW!