LowPowerLab/RFM69 #53: SPCR was not declared?

solars, 2016-11-01:
Hi there,

Trying to get started with RFM69, but if I try to compile a basic example I get:
https://gist.github.com/solars/1e52718dacacdb018de3efc62206d9f8

Any idea how to solve this?

Thanks a lot

LowPowerLab, 2016-11-01:
You need to do this from the ArduinoIDE, the SPCR is a core register defined in the AVR classes.

closed 2016-11-01 by LowPowerLab

solars, 2016-11-01:
How do you mean? I'm starting it in the arduino ide - but it's a wemos d1 mini

LowPowerLab, 2016-11-01:
What's your target board?

solars, 2016-11-01:
The wemos d1 mini

LowPowerLab, 2016-11-01:
Sorry but I don't know what that is of if that has a SPCR.

solars, 2016-11-01:
It's a ESP8266 board, not sure either: https://www.wemos.cc/product/d1-mini.html

LowPowerLab, 2016-11-01:
Oh ok, that won't have an SPCR.

solars, 2016-11-01:
Does this mean the lib won't work? And does this depend on the board? I read that in general the ESP8266 works. Thanks a lot for your help

LowPowerLab, 2016-11-01:
I know others use the ESP8266 with RFM69 using this library. You may want to check www.lowpowerlab.com/forum

LowPowerLab, 2016-11-02:
BTW the SPCR checks are wrapped in `#ifdefs`:
`#if defined (SPCR) && defined (SPSR)`
So you should never have a compile problem even if SPCR is not defined.