LowPowerLab/RFM69 #54: fatal error: avr/wdt.h: No such file or directory

Pixinger, 2016-11-22:
Hi
I'm not sure if it is really a problem, or not. I'm using Arduino 1.6.12 (Windows).
After downloading RFM69 and SPIFlash to the library and loading the "Node" example, I get the following error:

C:\Users\Pixinger\Documents\Arduino\libraries\RFM69\RFM69_OTA.cpp:35:21: fatal error: avr/wdt.h: No such file or directory
 #include <avr/wdt.h>

The weired thing is, it is physically available: C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include\avr\wdt.h

Is someone having similar problems?
Kind regards 
Pix

LowPowerLab, 2016-11-22:
Any other errors or just that is the only one?

LowPowerLab, 2016-11-22:
I just downloaded 1.6.12 and compiling Node and any other sketch has no issue.
Also this goes through [travis-ci](https://travis-ci.org/LowPowerLab/RFM69) and no issues with the build there either.

closed 2016-11-22 by LowPowerLab

Pixinger, 2016-11-22:
Thank you for the quick response. No I have no other issues or warnings. 
I'm using the Adafruit Feather M0 RFM69HCW 433. They write in their FAQ that missing header files may be due to incompatibility to the feather. 

[https://learn.adafruit.com/adafruit-feather-m0-radio-with-rfm69-packet-radio/power-management?view=all#faq
](url) (search for "Missing header files")

Maybe the feather M0 is just incompatible and has no watchdog support (I'm no PRO in that).
However, they (adafruit) mention also the "RadioHead" library. This library is working well, so it's not a big problem for me, if the feather is not compatible. It meight also be my fault... I don't know. 
If you would like to investigate this problem in more detail - I will do what I can. 

Kind regards 
Pix

Pixinger, 2016-11-22:
Just for your information.
It seems to be a problem of the "Adafruit Feather Cortex M0+ Radio Packet 433MHz".

If I switch the Arduino 1.6.12 project to a "Arduino/Genuino Uno" it compiles without any issues. 
If I switch back to "Adafruit Feather M0" the problems comes back.

Kind regards
Pix

LowPowerLab, 2016-11-22:
Yup, feather/M0 fail :-(
Thanks for the follow up.

lpercifield, 2016-11-22:
I'm working extensively with m0 and RFM69HCW and the lowpowerlab library. I
haven't tried the OTA, but that requires external flash as related to the
motino. I've used the library on sparkfun m0 dev board, adafruit feather m0
and custom m0 boards, as well as arduino ide 1.6.9 and 1.6.12. Happy to
help if you are still having issues.

LowPowerLab, 2016-11-22:
@lpercifield - did you try the latest RFM69 lib?
It was updated a few days ago with some significant changes.

lpercifield, 2016-11-22:
I haven't tried the newest version, I'll download it and see what happens.

Pixinger, 2016-11-23:
Thanks for both your responses. 
It would be good to hear the results Ipercifield has with the updated sources.

gfvalvo, 2016-11-27:
Hi,
I downloaded the RFM69 library yesterday and had the EXACT same problem when trying to compile with Adafruit Huzzah ESP8266 as the target board. Posted to the forum:
[https://lowpowerlab.com/forum/rf-range-antennas-rfm69-library/rfm69-radio-with-huzzah-esp8266/](url)

For now I've solved the issue by simply removing RFM69_OTA.cpp from the RFM69 library folder. My sketch wasn't using any of the functions provided by this file. It looks like the WDT is only used to reset the processor after an OTA code update. It hangs the processor in a tight loop until the WDT kicks in. The WDT in the ESP8266 processor is already setup to do this, I think that why it needs a yield() call in a tight loop. So, the wdt.h header file and wdt_enable(WDTO_15MS) function call aren't required for this processor, just enter the loop. Some MCU-dependent conditional compilation statements in this area would probably fix it.

Greg

LowPowerLab, 2016-11-28:
If you are able to provide guidance or a pull request I am willing to build-test this (no hardware test) and merge it in the library. Just not familiar with how this would be coded for the ESP. Thanks!

gfvalvo, 2016-11-28:
I'm sorry, I'm far too new to the "Arduino" ecosystem and my range of experience with ESP8266 boards is too limited (only Adafruit Feather Huzzah ESP8266) to feel comfortable making a recommendation for the generic ESP8266 universe. Plus, I have not yet explored any OTA functions with the RFM69 library.

Here's what I know -- on a Huzzah ESP8266 this loop:

`while(1) {
  ;
}
`

will very quickly cause a processor reset without any need for the wdt.h file or the wdt_enable() call. So, I'd conditionally omit those references for the Huzzah ESP8266.
I THINK the same holds true for all ESP8266-based board.

Thanks, nice library. I now have bi-directional 915 MHz communications going between Huzzah ESP8266 and Feather 32U4.

Greg

LowPowerLab, 2016-11-28:
Thanks, ok perhaps someone else could submit a PR and I can merge.

Pixinger, 2016-11-30:
@LowPowerLab: Sorry for getting back so delayed. 
Thanks @gfvalvo for your response. It settles me to hear that you get the same result. 
I would be happy to assist providing a patch or something, but unfortunately I'm also far to new to the arduino "ecosystem". To recommend patches, I think I need to understand better, what exactly is going on behind the scenes. Maybe the producer of the feather adafruit.com is interested in helping for a solution. I guess they should have the required knowledge. They provide a quick tutorial on their page and reference to the LowPowerLab project. 
If you want, I can try to contact them, although I don't know if they are interested/having time/ressources/ willing/possibilities to help.

lpercifield, 2016-12-06:
If I find some time, I hope I can help tackle this issue, but the usual
trick is the wrap any architecture dependent includes in defines. Arduino
requires an architecture define as part of the boards package, so it should
be relatively simple. There are plenty of other examples out there of this
technique if someone else beats me to it.