LowPowerLab/RFM69 #56: Adafruit M0

fabltd, 2016-12-01:
Hi

Adafruit give this example:

https://learn.adafruit.com/adafruit-feather-m0-radio-with-rfm69-packet-radio/power-management?view=all#using-the-rfm69-radio

This code will not compile:

RFM69-master\RFM69_OTA.cpp:35:21: fatal error: avr/wdt.h: No such file or directory

 #include <avr/wdt.h>

compilation terminated.

Does it work with an m0 or are they wrong?

LowPowerLab, 2016-12-01:
This library is written primarily for AVR not ARM.
If someone submits a PR after a fix is tested I can merge it.

fabltd, 2016-12-01:
Hi

Some one needs to tell Ada fruit. Its the example library they give it as
an example for Arm.

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

LowPowerLab, 2016-12-01:
Or you could use a Moteino to support the development of this free library :+1:

lpercifield, 2016-12-01:
Or if you don't care about being on the bleeding edge of the code, I have a fork that works for SAMD21 MCUs: https://github.com/lpercifield/RFM69/tree/SAMD21

fabltd, 2016-12-01:
@lpercifield  Will try that now.

lpercifield, 2016-12-01:
Just FYI, this branch includes a change that allows for 32bit node IDs, it changes the max payload a little. I did this so that the MCU can auto generate a "unique" node ID from its own UUID.

fabltd, 2016-12-01:
Excellent.

This works. It should be merged with the master :-)

It would have saved me loads of time

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

lpercifield, 2016-12-01:
I'll look at merging in the current LowPowerLabs master branch, but as it stands I don't think my branch would be a candidate to merge back upstream.

fabltd, 2016-12-01:
Hi

Its working so that's the main thing :-)

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

fabltd, 2016-12-06:
@lpercifield  Does your fork work with the ESP8226 ?

Thanks

lpercifield, 2016-12-06:
I'm not using the ESP chips in my current setup, but please report back if
you get it working.

fabltd, 2016-12-06:
Hi

Its working with ESP. However i am getting the following periodically.

Soft WDT reset

ctx: cont
sp: 3ffef290 end: 3ffef530 offset: 01b0

>>>stack>>>
3ffef440:  00000070 3ffee404 3ffee398 402020dd
3ffef450:  00000024 3ffee404 3ffee398 40202091
3ffef460:  3ffef4f0 00010741 3ffee398 402026f0
3ffef470:  0000003d 00010741 3ffee398 40202750
3ffef480:  3ffef4f0 3ffee398 3ffee398 4020296c
3ffef490:  0000000f 00000001 3ffef4fe 3ffee504
3ffef4a0:  3fffdad0 0000000f 3ffee4e0 3ffee504
3ffef4b0:  00000001 3ffee398 00000000 4020200d
3ffef4c0:  3ffef4f0 0000000f 00000002 00000028
3ffef4d0:  40201136 000003e8 3ffee4e0 3ffee504
3ffef4e0:  3fffdad0 3ffee398 3ffee4e0 40201e10
3ffef4f0:  6c6c6548 6f57206f 20646c72 00353323
3ffef500:  00000000 00000000 00000001 40203319
3ffef510:  3fffdad0 00000000 3ffee4fc 40203344
3ffef520:  feefeffe feefeffe 3ffee510 40100718
<<<stack<<<

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

LowPowerLab, 2016-12-06:
Great, thanks.
Not sure what that stack is though.

gfvalvo, 2016-12-07:
My quick and dirty solution for the /avr/wdt.h problem was to completely remove the RFM69_OTA.cpp file from my RFM69 library folder. I wasn't doing OTA firmware updates anyway.

This "fixed" the /avr/wdt.h  problem for ESP8266, however see here for another ESP8266 issue:
[https://github.com/LowPowerLab/RFM69/issues/49](url)

The fix suggested there helped greatly. I'll certainly look at LowPowLab product to help support library development. But, for my current project, the ESP8266's built-in WiFi made it a no-brainer.

Greg

fabltd, 2016-12-08:
@LowPowerLab 

Sorry quick question. radio.DATA is the incoming buffer what data type is this. 

I am trying to send this out via serial as bytes but i am failing.

LowPowerLab, 2016-12-08:
yes, byte buffer, see examples on usage.

MPParsley, 2017-04-01:
I tested https://github.com/lpercifield/RFM69/tree/SAMD21 and it seems to have fixed this error: "RFM69_OTA.cpp:35:21: fatal error: avr/wdt.h: No such file or directory".

This branch seems to have deleted RFM69_OTA all together. Perhaps we can create a PR that disables OTA for non AVR hardware?

LowPowerLab, 2017-04-03:
You're welcome to submit a PR for that.

joymonkey, 2017-11-15:
I can't create PR's here (for some reason the "New pull request" button just brings me to https://github.com/LowPowerLab/RFM69/compare ), but I'd love an option to disable RFM69_OTA too.

LowPowerLab, 2018-07-16:
Solved with release of Moteino M0.

closed 2018-07-16 by LowPowerLab