LowPowerLab/RFM69 #57: ESP8226

fabltd, 2016-12-06:
Does this library work with the ESP8266?

LowPowerLab, 2016-12-06:
It does or it used to.

closed 2016-12-06 by LowPowerLab

fabltd, 2016-12-06:
Used to?

LowPowerLab, 2016-12-06:
Yes it used to work but changes were made and the CI does not run for ESP so I can't guarantee it still works. This lib is primarily created and supported for AVR.
Please post any issues in the http://lowpowerlab.com/forum, and only open an issue if there is one.
Else you can submit a PR to contribute support for other platforms.
Thanks

fabltd, 2016-12-06:
complains about avr/wdt.h ?

Is this available as a library.

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

LowPowerLab, 2016-12-06:
wdt is an AVR library.
ESP might have a problem with that so that part of the code might need to be wrapped in defines if ESP does't use it.

rrobinet, 2016-12-08:
yes 
See https://lowpowerlab.com/forum/moteino/moteino-w5100-ethernet-spi-support-spi_has_transaction/msg15583/#msg15583

rrobinet, 2016-12-08:
FYI 
RFM69 library is fully compatible with ESP8266 ( however the last version including OTA, may not, because ESP8266 has its own OTA programming!).  There are some programming differences with Arduino /Moteino, due to the architecture itself (typically EEPROM),  some Arduino (C++)  functions are not implemented (typically sscan), data structure has some differences (typically integer), etc.. 
Also ESP8266 doesn't like  loops longer than 2 seconds, because it has to take care of WiFi background processing, so a long loop  causes a watchdog reset.  yield() or delay(0) is to be used to allow background processing in long loops.
For RFM69 and ESP8266 compatibility, see  also my note:
https://lowpowerlab.com/forum/moteino/moteino-w5100-ethernet-spi-support-spi_has_transaction/msg15583/#msg15583

fabltd, 2016-12-08:
> ethernet-spi-support-spi_has_transaction/msg15583/#msg15583

Hi Robert

Does the loop issue  explain this  stack error?

*Decoding 12 results*0x402020dd: *RFM69::writeReg(unsigned char,
unsigned char)* at Arduino\libraries\RFM69-SAMD21/*RFM69.cpp* line
*205*0x40202091: *RFM69::readReg(unsigned char)* at
Arduino\libraries\RFM69-SAMD21/*RFM69.cpp* line *205*0x402026f0:
*RFM69::readRSSI(bool)* at Arduino\libraries\RFM69-SAMD21/*RFM69.cpp*
line *449*0x40202750: *RFM69::canSend()* at
Arduino\libraries\RFM69-SAMD21/*RFM69.cpp* line *223 (discriminator
1)*0x4020296c: *RFM69::send(unsigned int, void const*, unsigned char,
bool)* at Arduino\libraries\RFM69-SAMD21/*RFM69.cpp* line *235
(discriminator 1)*0x40201f28: *RFM69::receiveDone()* at
Arduino\libraries\RFM69-SAMD21/*RFM69.cpp* line *205*0x4020200d:
*RFM69::sendWithRetry(unsigned int, void const*, unsigned char,
unsigned char, unsigned char)* at
Arduino\libraries\RFM69-SAMD21/*RFM69.cpp* line *205*0x40201136:
*delay* at Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/*core_esp8266_wiring.c*
line *53*0x40201e10: *loop* at Arduino\esp8226radio/*esp8226radio.ino*
line *75*0x40203319: *esp_schedule* at
Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/*core_esp8266_main.cpp*
line *56*0x40203344: *loop_wrapper* at
Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/*core_esp8266_main.cpp*
line *56*0x40100718: *cont_norm* at
Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/*cont.S*
line *109*

-- 
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel:       +44 (0)1727 238 894
Twitter: @fabcontrols

                              DALI Lighting /Emergency test solution
providers
                     Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
                                         St Albans  . Hertfordshire . AL3
6EN

fabltd, 2016-12-08:
@rrobinet 

I note your changes here. 

https://github.com/rrobinet/RFM69_Libary

Do I just overwrite the existing files?

rrobinet, 2016-12-08:
Hi Jeremy,
No,
Issue with watchdog reset are clearly identified with a registered dump and a WDT reason, while the code is running.
Something like:
Soft WDT reset

ctx: cont
sp: 3ffef600 end: 3ffef890 offset: 01b0

>>>stack>>>
3ffef7b0: 3ffee676 00000004 3ffee65c 402022e5
3ffef7c0: 000003e7 3ffee65c 0000e7ad 40202492
3ffef7d0: 3ffee650 00000001 00000040 4020232c
3ffef7e0: 3ffee650 00000001 3ffee65c 000003e7
3ffef7f0: 3ffee650 00000001 3ffee65c 40202b19
3ffef800: 0000e7ad 0000000c 00000001 40202074
3ffef810: 00000009 3ffee65c 00000000 0000e785
3ffef820: 00000001 3ffee65c 00000002 40202144
3ffef830: 3ffee650 0000000c 00000002 00000028
3ffef840: 33333420 7a684d20 3ffee7ac 000000be
3ffef850: 3ffee650 3ffee65c 3ffee7ac 40201f4a
3ffef860: feefeffe 00000000 00000000 3ffee868
3ffef870: 3fffdad0 00000000 3ffee860 40203e70
3ffef880: feefeffe feefeffe 3ffee870 40100718
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

 I am using the my version of RFM69 library https://github.com/rrobinet/RFM69_Libary  compatible with Moteino+ Ethernet and ESP8266 on an native ESP8266 ESP-12 and WeMos without particular issues, beside programming particularities

This with Arduino IDE 1.6.12.  and the ESP8266 package 2.3.0 on MAC OSX

The dump you are talking about looks to me a IDE compilation issue.

Robert

--
Regards,

Jeremy Turner
Managing Director
FAB Controls Ltd
Mobile: +44 (0)797 131 4661
Tel: +44 (0)1727 238 894
Twitter: @fabcontrols

DALI Lighting /Emergency test solution
providers
Unit 6 . St Albans Enterprise Centre . Long Spring .
Porters Wood
St Albans . Hertfordshire . AL3
6EN

rrobinet, 2016-12-08:
Yes,
This one of the previous version that do not include the OTA part, you can still use the examples

https://github.com/rrobinet/RFM69_Libary