LowPowerLab/RFM69 #58: ack timeout when a broadcast is sent ...

voidptr69, 2016-12-07:
In my case I need to use send with retry all the time for direct address and broadcast. 
of course in broadcast I should loose the retry and ack functionalities.

I think you try to catch this in the ACKReceived  but it is not working fot a braoadcast because my receivers will not ack a broadcast message.

so the sendWithRetry will spin and time out...

here a dirty modification to ACKReceived that exit with true immediatly went a broadcast address was used, so not time out.

:-)

bool RFM69::ACKReceived(uint8_t fromNodeID) {
  
  if(fromNodeID == RF69_BROADCAST_ADDR) {
    receiveDone();
    return true;
  }
  if (receiveDone()) {
    return( ((SENDERID == fromNodeID) && ACK_RECEIVED ));
    //return (SENDERID == fromNodeID || fromNodeID == RF69_BROADCAST_ADDR) && ACK_RECEIVED;
  } 
  return false;
}

bool RFM69::sendWithRetry(uint8_t toAddress, const void* buffer, uint8_t bufferSize, uint8_t retries, uint8_t retryWaitTime) {
  uint32_t sentTime;
  for (uint8_t i = 0; i <= retries; i++)
  {
    send(toAddress, buffer, bufferSize, true);
    sentTime = millis();
    while (millis() - sentTime < retryWaitTime)
    {
      if (ACKReceived(toAddress))
      {
        Serial.print(" ~ms:"); Serial.print(millis() - sentTime);
        return true;
      }
    }
    Serial.print(" RETRY#"); Serial.println(i + 1);
  }
  return false;
}

// check whether an ACK was requested in the last received packet (non-broadcasted packet)
bool RFM69::ACKRequested() {
  return ACK_REQUESTED && (TARGETID != RF69_BROADCAST_ADDR);
}

LowPowerLab, 2016-12-07:
A broadcast should never contain an ACKREQUEST.
You should not use sendWithRetry with the BROADCAST address.

closed 2016-12-07 by LowPowerLab

voidptr69, 2016-12-08:
I think I know that, I was just telling you that your lib can be a bit more generixc....

so then why are you chechking for broadcast address in bool RFM69::ACKReceived  :-)