LowPowerLab/RFM69 #61: setPowerLevel sets the wrong level for HW radios

gpshead, 2017-01-16:
Datasheet section 6.3 Transmitter Registers, Table 25 has a footnote regarding OutputPower saying "Only the 16 upper values of OutputPower are accessible" when any of PA1 or PA2 are on.

the existing code has an `if (_isRFM69HW) _powerLevel /= 2;` which seems wrong as it'll force the supplied 0..31 value into the 0..15 range rather than in the 16..31 range as the datasheet implies is intended with its OutputPower -> dBm calculations. I suggest adding 16 after you divide by 2 so that your 0..31 input maintains its meaning and is spread across the documented allowed register value range.

The actual behavior of my RFM69CHW in high power mode appears similar regardless of which way I write the register (but I don't have a scope, just watching the RSSI on another RFM69 receiver as I play with a transmitter). So it may have internal logic that works either way. But what I describe above is consistent with the -11dBm + OutputPower = tx dBm description in the datasheet.

LowPowerLab, 2017-01-16:
Actually this is intentional specifically because the SX1231 divides the range in half when the upper PAs are used (in the case of RFM69 HW/HCW).

mje-nz, 2018-04-19:
Sorry to resurrect this issue, but I came here to ask the same thing and I don't understand your response. Isn't the datasheet (both for RFM69HW and for SX1231) pretty clear that when PA1 or PA2 is enabled, the only valid output powers are 16-31?

LowPowerLab, 2018-04-19:
Yes, that is why the library uses increments of 2.

mje-nz, 2018-04-19:
Are you saying that this function sets the OutputPower field to values in [16, 31]?

```
void RFM69::setPowerLevel(uint8_t powerLevel)
{
  _powerLevel = (powerLevel > 31 ? 31 : powerLevel);
  if (_isRFM69HW) _powerLevel /= 2;
  writeReg(REG_PALEVEL, (readReg(REG_PALEVEL) & 0xE0) | _powerLevel);
}
```

LowPowerLab, 2018-04-20:
[0,31] for W/CW
[0,15] for HW/HCW: `if (_isRFM69HW) _powerLevel /= 2`

mje-nz, 2018-04-21:
Ok, so you agree that the code doesn't follow what the datasheet says to do then? What was your objection to @gpshead's suggestion of adding 16?

kowalej, 2019-12-16:
Was this ever addressed? Seems to me that you should use 16 - 31 versus the current 0 - 15 levels for HW/HCW radios.

LowPowerLab, 2019-12-16:
0-31 is standard for all radios. Abstracts the differences between CW/W and HCW/HW. The HW/HCW are higher power (level 31 corresponds to their max power of 20dBm), the W/CW are lower power (level 31 corresponds to max output of 13dBm).

jwjulien, 2021-06-07:
I just stumbled upon this and found this issue.  I realize this is old but I think it's still an issue.

I agree with @mje-nz on this - the value written to the device, regardless of the model number, should be [0-31] as it's the least-significant bit that is ignored by the device.  The code as it stands eliminates the most significant bit in the power range effectively halving the maximum output power for the "high power" radios.

If you want to keep the special case in the code for the `_isRFM69HW` case then might I propose:
```
void RFM69::setPowerLevel(uint8_t powerLevel)
{

  _powerLevel = (powerLevel > 31 ? 31 : powerLevel);
  if (_isRFM69HW) _powerLevel &= 0x1E;
  writeReg(REG_PALEVEL, (readReg(REG_PALEVEL) & 0xE0) | _powerLevel);
}
```

Though, admittedly, it seems inconsequential to even cater to the "H" devices - the library could just ignore the special case and let the radio ignore the least significant bit.

ygoe, 2021-08-24:
First, the original issue never seemed to be addressed here. The author always responded with a confusing statement, never with a clear and concise explanation of why his statement should even be correct. The RFM69 datasheet doesn't contain any of this and the SX1231 datasheet is clearly not implemented by the code. Does it work anyway? I don't know.

Second, I find the power level code very convoluted. It seems like I can only enable PA1 and PA2 together, but not individually. The code even states that both have to be enabled for the H models for anything to work. I cannot conclude that from the datasheet. I'm thinking about a new function that lets me specify the desired output dB and then determines the correct registers to set on its own. But it's hard to achieve with the current code architecture because some registers overlap in features and have to be changed between sending/receiving.

The SX1231 datasheet adds to my confusion. It only mentions a range up to 17 dBm. 13 dBm can be used with PA0 or PA1 alone and 17 dBm can be used with PA1 and PA2 combined. The SX1231 and HopeRF RFM69 datasheets massively differ here. One says PA1 goes from -18 to +13 (which renders PA1 useless because it's the same as PA0) and the other says PA1 goes from -2 to +13 (max is the same but min is much higher, no idea why). Which is true? Both show different tables of different output ranges with different PA settings. It's all a big mess and it's probably impossible to use these chips in any deterministic way without reverse-engineering it all with expensive measuring equipment. This library isn't of any use for setting the output power in a specific way.

LowPowerLab, 2021-08-24:
@ygoe take a deep breath and relax ;-)

The RFM69 HW/HCW (20dBm or 100mW output) are wired differently than the CW (13dBm), and they make use of the PA amps differently yielding the different power outputs. How that is I will leave as homework for you to determine from the RFM69 schematics, not sure how useful it is for me to spend an hour digging up the details and offer the `demanded concise explanation` since it sounds like you already flushed this library and moved on. Here's my other 2 cents:

The SX1231 is the datasheet of the transceiver chip. The RFM69 datasheet is the HopeRF butchery of that. After you realize these 2 truisms, it's like a major revelation.

I thought [0-31] is much easier to understand than ranges of dBs like [-2..13]dBm or [5..20]dBm. But apparently not all folks see the light the same way I do. The power levels are also very useful in the ATC extension, not so much db's since we have no clue how to `deterministically` increase power according to perceived power at a far end node. Plus, what if I want half or 2/3 of a dB? I can easily express that in math, but then if the register has a resolution of only 5 bits ... the effort is quite pointless.

The library is made as lean as possible, hence it power takes levels which directly translate into values in the chip registers. If you want to convert from a linear scale into logarithmic/dBs because only that is acceptable, you are welcome to write a function to determine that according to the datasheet ecuations and pass that into the setPowerLevel().

But it sounds like you already decided to move on so I won't bother you too much with more explanations.

ygoe, 2021-08-24:
In another issue here I found a link to a [blog post that investigates the power level settings](https://andrehessling.de/2015/02/07/figuring-out-the-power-level-settings-of-hoperfs-rfm69-hwhcw-modules/) and their effects more in-depth. Its conclusion for the H models is to stay with PA1 and only add PA2 and then the boost option for higher levels. There's another link in the comments there that [explores the differences between non-H and H models](https://www.electrobob.com/rfm69-output-power/) and even provides C code to set the PA level register from the dBm value in the recommended way. That was exactly the kind of code I had imagined there. It needs some integration with this library to properly update the PA1/2 settings when switching between TX and RX. A class variable should do to remember the intended values.

Using the dBm value as intent instead of a shifting unitless scale should always be better because 1. it's more obvious what's happening and 2. it allows scaling up or down the transmit power automatically (for example for retries) without thinking about every hardware implementation detail where such consideration is not appropriate (think of clean code, a concept that doesn't seem to have reached the hardware world yet).

LowPowerLab, 2021-08-24:
@ygoe IMHO#1 [0-31] is more sensical to a broader user base, and it abstracts the hardware implementation. The ATC extension actually automatically ++/-- adjusts the power level based on perceived RSSI at the far end, "without thinking of every hardware implementation".
IMHO#2 a better (and possibly cheaper) way of measuring transmitted power is hooking up a [current measuring device](https://lowpowerlab.com/shop/product/152) to a scope (shameless plug, sorry), it is far easier and more intuitive than the setup in the linked video, and more useful than showing just transmitted power that comes out of an antenna.

I am curious though, about your measurements, regardless of the method you might be using.

ygoe, 2021-08-24:
The thing is, [0-31] isn't even working. It gives the impression that there are 32 increasing power levels where there are actually 22 and this single value alone isn't enough to address all of them. (Actually the code locks me to PA1+PA2+boost.) To use the full spectrum of power levels, and energy-efficiently so, the PA and boost settings (3 more flags to the 16 usable levels) must be used (wisely). That's complicated and requires a lot of intimate hardware knowledge. I still believe that a function that abstracts all of that away is more useful. It's probably irrelevant that it maps to real dBms here (from -2 to 20), but still nice. The implementation could also map this to [0, 22] to avoid negative numbers, if that's important. But then it wouldn't compare with the non-H models anymore or they'd need another mapping like the current [0, 31].

Yes, the video uses a complex setup. I just hooked up the probe at the foot of my antenna (wire soldered to the board, standing straight up).

LowPowerLab, 2021-08-30:
@ygoe I spent the last few days changing the library and testing these changes. I agree the setPowerLevel() function was a bit smelly. I did actual power measurements with various nodes and I believe I have arrived at a good stopping point. In the process I discovered some other bugs as well. This change is now also extended to RFM69_ATC which was causing the power levels to change in unexpected ways so this extension in particular is significantly refactored and optimized now. I [included 2 new examples](https://github.com/LowPowerLab/RFM69/commit/07bdb458d5817dc6b7bd6ba3aa55b86d2f8c4b10) that make it easy to reproduce the power measurements I've done with a [CurrentRanger](https://github.com/LowPowerLab/currentRanger/) and a receiver which can demonstrate the output power increasing and decreasing. I will make a new release v1.5 to include all this. The HW/HCW radios now have a 24 step range and W/CW have 32 steps - I wanted to keep it consistent but after extensive testing and attempts at having the smoothest possible increase in power and current I narrowed it down to 24 levels for HW/HCW, for W/CW things are more simple since there is only 1 PA available - PA0 and there are 32 distinct levels that output power in a linear way so why not use the whole range. I will hopefully write a blog explaining all this. I also added a setPowerDBm() function similar to the @ahessling [RFM69-STM32 library](https://github.com/ahessling/RFM69-STM32/blob/master/rfm69.cpp#L815).
Anyway thanks for bugging me, sometimes I need a little of that to distract me from my other projects and dive in to fix some things.

closed 2021-08-30 by LowPowerLab

ygoe, 2021-08-31:
Thanks for the fix, and you're welcome for the bugging. ;-)

LowPowerLab, 2021-08-31:
@ygoe, @jwjulien @mje-nz, @kowalej, @gpshead - thanks for the very long patience on this issue. I believe it is now fixed and I [posted a blog about my findings](https://lowpowerlab.com/2021/08/31/rfm69-tx-output-power-testing-library-fix/). Please try it out (and try the new `setPowerDBm()` function) and let me know if you spot anything amiss.

commit b4f3f11b (2021-08-30): SetPowerLevel() fix & ATC refactoring

Fixes #61