LowPowerLab/RFM69 #65: readReg(0x30) (networkID) returns 0, albeit set something else

eroniki, 2017-02-06:
Hello,

I have been struggling with my RFM69HCW modules for a while. Even though the transmitting side seems to be working, I cannot receive anything at the other end of the transmission. I started investigating the issue by reading the registers. This is how I realized something interesting. 

In details, as far as I understand the register 0x30 of the module is used for network id which set during the initialization process. In theory, when the register is read after initialization I should be reading the value set during initialization. However, when I read the value with the function `readReg(0x30)` I receive 0 no matter what value was assigned during the initialization. 

I would be glad if you let me know if it is problem with my modules or I am reading the wrong register.

I am providing a hunk of the code running the nodes below.

Thanks in advance.

Bests,
M

```Arduino
#include <RFM69.h>    //get it here: https://www.github.com/lowpowerlab/rfm69
#include <SPI.h>

#define NETWORKID     6  // The same on all nodes that talk to each other
#define NODEID        99    // The unique identifier of this node
#define FREQUENCY     RF69_915MHZ
#define IS_RFM69HCW   true // set to 'true' if you are using an RFM69HCW module
#define ENCRYPTKEY     "sampleEncryptKey" //exactly the same 16 characters/bytes on all nodes!
readReg(0x30) 
//*********************************************************************************************
#define SERIAL_BAUD   115200

#define RFM69_CS      10
#define RFM69_IRQ     2
#define RFM69_IRQN    0  // Pin 2 is IRQ 0!
#define RFM69_RST     9

RFM69 radio = RFM69(RFM69_CS, RFM69_IRQ, IS_RFM69HCW, RFM69_IRQN);

int16_t packetnum = 0;  // packet counter, we increment per xmission
int16_t rssi[8];

void setup() {
  while (!Serial); // wait until serial console is open, remove if not tethered to computer
  Serial.begin(SERIAL_BAUD);

  Serial.println("Arduino RFM69HCW Transmitter");

  delay(100);

  // Initialize radio
  radio.initialize(FREQUENCY, NODEID, NETWORKID);
  if (IS_RFM69HCW) {
    radio.setHighPower();    // Only for RFM69HCW & HW!
  }
  radio.setPowerLevel(31); // power output ranges from 0 
  // Hard Reset the RFM module
  pinMode(RFM69_RST, OUTPUT);
  digitalWrite(RFM69_RST, HIGH);
  delay(100);
  digitalWrite(RFM69_RST, LOW); //(5dBm) to 31 (20dBm)

  radio.encrypt(ENCRYPTKEY);

  Serial.print("\nTransmitting at ");
  Serial.print(FREQUENCY == RF69_433MHZ ? 433 : FREQUENCY == RF69_868MHZ ? 868 : 915);
  Serial.println(" MHz");
}
```

eroniki, 2017-02-06:
I also realize that I usually obtain 0 with `getFrequency()` as well.

LowPowerLab, 2017-02-06:
The library is tested on Moteinos and the provided examples are tested on various applications for this platform, without the need to write anything else to the registers (except in expert mode where you know what exactly you are doing). One issue is you may have faulty modules. Please consult the examples once you are certain the modules are good and wired correctly to a Moteino or supported AVR.

closed 2017-02-06 by LowPowerLab