LowPowerLab/RFM69 #66: CLKOUT defines are not correct

boaschti, 2017-02-07:
Hi,

In registers.h

#define RF_DIOMAPPING2_CLKOUT_32          0x00
#define RF_DIOMAPPING2_CLKOUT_16          0x01
#define RF_DIOMAPPING2_CLKOUT_8           0x02
#define RF_DIOMAPPING2_CLKOUT_4           0x03
#define RF_DIOMAPPING2_CLKOUT_2           0x04
#define RF_DIOMAPPING2_CLKOUT_1           0x05  // Reset value
#define RF_DIOMAPPING2_CLKOUT_RC          0x06
#define RF_DIOMAPPING2_CLKOUT_OFF         0x07  // Recommended default

I think it's wrong, it must be:

#define RF_DIOMAPPING2_CLKOUT_1          0x00
#define RF_DIOMAPPING2_CLKOUT_2          0x01
#define RF_DIOMAPPING2_CLKOUT_4          0x02
#define RF_DIOMAPPING2_CLKOUT_8           0x03
#define RF_DIOMAPPING2_CLKOUT_16           0x04
#define RF_DIOMAPPING2_CLKOUT_32           0x05 
#define RF_DIOMAPPING2_CLKOUT_RC          0x06  // Reset value
#define RF_DIOMAPPING2_CLKOUT_OFF         0x07  // Recommended default

Best regards

boaschti, 2017-02-07:
Or:

#define RF_DIOMAPPING2_CLKOUT_32Mhz 0x00
#define RF_DIOMAPPING2_CLKOUT_16Mhz 0x01
#define RF_DIOMAPPING2_CLKOUT_8Mhz 0x02
#define RF_DIOMAPPING2_CLKOUT_4Mhz 0x03
#define RF_DIOMAPPING2_CLKOUT_2Mhz 0x04
#define RF_DIOMAPPING2_CLKOUT_1Mhz 0x05 
#define RF_DIOMAPPING2_CLKOUT_RC 0x06  // Reset value
#define RF_DIOMAPPING2_CLKOUT_OFF 0x07 // Recommended default
:-/

LowPowerLab, 2017-02-08:
Thanks. Actually the definitions are correct but the naming maybe is a not obvious.
For instance `#define RF_DIOMAPPING2_CLKOUT_32          0x00` refers to 32mhz not division by 32, and this matches the datasheet. I may change the names of these to reflect that.
See [DS page 69](http://www.semtech.com/images/datasheet/sx1231h.pdf) RegDioMapping2
 definitions in section 6.5.

closed 2017-02-08 by LowPowerLab