LowPowerLab/RFM69 #74: On board GPIO

fabltd, 2017-04-16:
How do you access the GPIO outs from the RFM69HCW PCB ?

LowPowerLab, 2017-04-17:
DIO0 is connected to D2 (INT0) interrupt pin for use in the library.
The others you will have to connect directly from the RFM module.
The MoteinoMEGA has jumper options to connect DIO1 and DIO2 and RST.

closed 2017-04-17 by LowPowerLab

fabltd, 2017-04-17:
Hi

Do you support the other gpio pins.

I want to add a dip switch for setting the network ids.

Thanks

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661

or mute
the thread

LowPowerLab, 2017-04-17:
That doesn't have much to do with the library. Do that on your MCU GPIO.

fabltd, 2017-04-17:
Hi

I am short on pins so hoped I could use the extra ones on the rfm69 .

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661

LowPowerLab, 2017-04-17:
The radio radio pins are not for general use but they have specific functions. See the sx1231h datasheet.
You can use a MoteinoMEGA for extra pins. BTW the analog pins on any Moteino can also be used as digital pins.

fabltd, 2017-04-17:
Thanks

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661