LowPowerLab/RFM69 #76: chaining the rfm on the spi bus

hjw, 2017-04-27:
I can successfully use this library, receive data on the radio and print it to the serial port but when I add another component to the spi chain the radio no longer can connect with it's partner. There is no SS conflict, the radio's on pin 10 and the other component on pin 5 (quadruple checked). 

I have searched with no success for any example of people pairing the radio with other spi components; I've only found projects with I2C sensors and then the radio on the spi bus. 

Have you ever successfully added another component to the spi chain along with the radio (ie an OLED display)?

TomWS1, 2017-04-27:
You'll probably have more response from the lowpowerlab forum, but I seem to recall there being an issue with the graphics library sharing SPI.  Like it kept the display's SS active more often than it should...  But I can't say for sure other than I've had many SPI devices (Moteino uses SPI flash) on the same bus without issue.

LowPowerLab, 2017-04-27:
@hjw It's most likely a conflict of libraries that do not handle the SPI bus properly.
See the [MightyHat example](https://github.com/LowPowerLab/RFM69/blob/master/Examples/MightyHat/MightyHat.ino), that uses a Nokia 5110 LCD in SPI mode, and SPI FLASH_MEM (same as on Moteino).

hjw, 2017-04-27:
OK, thanks.
I’d already suspected the SS being left active as a place to look but paid more attention to the radio end of it… I’ll look at the oled library’s end.

-Hawley

closed 2017-04-27 by LowPowerLab

rrobinet, 2017-04-29:
Hawley,

Please check:
https://lowpowerlab.com/forum/moteino/moteino-w5100-ethernet-spi-support-spi_has_transaction/msg15583/#msg15583
 It looks to be a SPI Interrupt  conflict problem as the one with the W5100 Ethernet Chip controller.
You should try my library version:
https://github.com/rrobinet/RFM69_Libary

Robert

hjw, 2017-04-29:
Thanks Rob. 
That thread has some good info. 
I will take a look at your version of the library and see if it solves my problem. 

-hawley