LowPowerLab/RFM69 #77: How FHSS work on RFM69?

AdriWicaksono, 2017-07-08:
Good Evening.

I'm Adriel Praba Wicaksono from Electronic Engineering Polytechnic Institute of Surabaya, Indonesia. I'm Telecommunication Engineering Final Year Student.

This year, I develop my final project. The title is "Wireless Communication System with Frequency Hopping Spread Spectrum using Raspberry Pi and FSK Modem". And i used RFM69HW as FSK Modem. But, i have some problem with the library. The problem is how to activate Frequency Hopping at RFM69HW. Is the FHSS activated by default in RFM69HW? How many frequency that synthesize by this module? Or there's unpredictable how many frequency that synthesize?  

Would you help me to tell me where's the source code parts/datasheet parts that explain it?

Thank You.

LowPowerLab, 2017-07-10:
Hi Adriel,
FHSS is not implemented in this library. You will have to use setFrequency() to change the channel according to a pseudorandom sequence. For more discussions, please do not open an issue here, but please use [the forum](https://lowpowerlab.com/forum).

closed 2017-07-10 by LowPowerLab

AdriWicaksono, 2017-07-17:
how to look the content of setFrequency()? 
Would you help me?
Thank's a lot.