LowPowerLab/RFM69 #87: Hardware address filtering

Imlerih, 2017-11-09:
Hello. Im trying to enable hardware address filtering, because in your lib its commented by default. The problem is when you have several nodes in one network each node will interrupt the MCU whatever nodeID it has, and only MCU decides if this message belongs or not. So I`d like to know how to make RMF intrerrupt MCU only if networkID AND nodeID matches.

LowPowerLab, 2017-11-09:
The library does not implement hardware addressing for reasons and limitations discussed in the forum. This is by design intentional, it's not a bug.
You can change some registers if you'd like and force hardware filtering, for that please see datasheet and also the forum might be helpful.

closed 2017-11-09 by LowPowerLab

Imlerih, 2017-11-09:
I noticed your comments and I understood, thet It wasn`t a bug. I tried to uncomment your comment and to add one more line to make it work, but it didn`t affect as I expected.

/* 0x39 */ { REG_NODEADRS, nodeID }, // turned off because we're not using address filtering
/* 0x3A */ { REG_BROADCASTADRS, RF69_BROADCAST_ADDR },

I read through datasheet, but didn`t finde any special commands, that would enable this function.