LowPowerLab/RFM69 #89: Trasmission time

Slartitorto, 2017-12-10:
Hi, thank very much for you great job, this library is very useful.
I figure out the transmission time is about 1 sec. but i think that time seems too much. Maybe I've wrong in my sketch ?
The receiver see the packet as soon the led in the transmitter node became on, so there is time not useful on the transmitter maybe awaiting the and of something or some timeout ?
Thanks in advance.

```
#include <LowPower.h>
#include <SPI.h>
#include <RFM69.h>

#define ledPin 3

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // Trasmitter node ID
#define TONODEID      1   // Destination node ID
#define FREQUENCY   RF69_433MHZ

RFM69 radio;

// -------------------------

void setup() {
  pinMode(ledPin, OUTPUT);

  Serial.begin(57600);

  radio.initialize(FREQUENCY, MYNODEID, NETWORKID);
  radio.setHighPower();

}

// -------------------------

void loop() {
   delay(10);

   char buff[] = "123 ABCDEFGHIJKLMNOPQRSTUVWXYZ";
   byte buffLen=strlen(buff);

   long start_timer=millis();
   
   digitalWrite(ledPin, HIGH);
   Serial.println("Sending buffer ...");
 
   radio.send(TONODEID, buff, buffLen);
   
   digitalWrite(ledPin, LOW);
   Serial.print("Trasmission time = "); Serial.println((millis() - start_timer));
   Serial.println();

   
   delay(200);
}

```

The output is:

```
Sending buffer ...
Trasmission time = 1014

Sending buffer ...
Trasmission time = 1012

Sending buffer ...
Trasmission time = 1014

Sending buffer ...
Trasmission time = 1013

Sending buffer ...
Trasmission time = 1012

Sending buffer ...
Trasmission time = 1014

Sending buffer ...
Trasmission time = 1013

Sending buffer ...
Trasmission time = 1012

Sending buffer ...
Trasmission time = 1014

Sending buffer ...
```

JohnOH, 2017-12-10:
You appear to be including the time for the Serial.print to complete in your calculation.

Slartitorto, 2017-12-10:
The Serial.print take some microsec to be execute.
However, I've take your suggest, trying this sketch, but the result doesn't change.

```
#include <LowPower.h>
#include <SPI.h>
#include <RFM69.h>

#define ledPin 3

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // Trasmitter node ID
#define TONODEID      1   // Destination node ID
#define FREQUENCY   RF69_433MHZ

RFM69 radio;

// -------------------------

void setup() {
  pinMode(ledPin, OUTPUT);

  Serial.begin(57600);

  radio.initialize(FREQUENCY, MYNODEID, NETWORKID);
  radio.setHighPower();

}

// -------------------------

void loop() {
   delay(10);

   char buff[] = "123 ABCDEFGHIJKLMNOPQRSTUVWXYZ";
   byte buffLen=strlen(buff);

   
   Serial.println("Sending buffer ...");
   digitalWrite(ledPin, HIGH);

   long start_timer=millis();
   radio.send(TONODEID, buff, buffLen);
   long end_timer=millis();

   digitalWrite(ledPin, LOW);
   Serial.print("Trasmission time = ");
   Serial.println(end_timer - start_timer);
   Serial.println();

   
   delay(200);
}
```

JohnOH, 2017-12-10:
I'm sorry it didn't help.

Slartitorto, 2017-12-10:
Solved: wiring problem.
My atmega328 had a cold welding on PD2.

closed 2017-12-10 by Slartitorto