LowPowerLab/RFM69 #108: RF69_IRQ_PIN for ATmega1284P

ljwinkler, 2018-06-17:
The RF69_IRQ_PIN 2 definition in RFM69/RFM69.h is incorrect.
According to a datasheet it is pin 16 (Arduino pin 10).

So the correct definition for:
`#elif defined(__AVR_ATmega644P__) || defined(__AVR_ATmega1284P__)`
is:
`
#define RF69_IRQ_PIN 10
`

Regards,
LJ

LowPowerLab, 2018-06-18:
Thanks but exactly what datasheet is this coming from?
The `#define RF69_IRQ_PIN          2` refers to the interrupt number not the digital pin.
The digital pin is obtained with the `#define digitalPinToInterrupt(p)` macro.

closed 2018-06-18 by LowPowerLab

ljwinkler, 2018-06-18:
Oh, got ya. However, I had to define it to 10 to get the display working. Haven't investigated it deeply. Checked that on 328 is 2 (I guessed it is digital pin 2) so I changed to 10 for 1284 and... it worked :)
Regards,
LJ

LowPowerLab, 2018-06-18:
If you really think this might be a bug you're welcome to share more details and I will look into it although I have to say nobody else reported a problem.
The digital pin 2 happens to match interrupt 2 on MoteinoMEGA and there is no connection to pin 10 at all.