LowPowerLab/RFM69 #109: Interrupt handler is slow

stevemk14ebr, 2018-06-25:
The interrupt handler routine is slow to execute and knocks other time sensitive interrupts off pace (as the cant execute while rfm interrupt fires). This is super bad if for example you want to control a motor and generate a pwm signal with another timer, as that signal will now be interrupted and you will get 'hitches' in the motors output. To fix i changed the interrupt handler to just set a flag, and dispatch the appriate logic on a callable function from 'loop': https://github.com/stevemk14ebr/RFM69

I opted not to do a PR since this is an interface breakage but it should still be noted imho

joysfera, 2018-07-28:
FYI, there's a simple solution for the same idea in https://github.com/andrei-ivanov/RFM69 - no need to call a new function from loop(), simply use the existing receiveDone(). Thus API is preserved but long time in IRQ handler is gone.

LowPowerLab, 2018-08-08:
Thanks guys, I pushed a new commit with the changes from this fork:
https://github.com/andrei-ivanov/RFM69/commit/2793a5a90981a963ca45b4e05c476b6e7139038b
Seems to work well so far. I will do more testing and if everything is OK I will make a new release.
For now please help test this change using latest RFM69-master and report any issues back in this thread. I will close it for now, will reopen if anyone reports a problem.
Thanks again

closed 2018-08-08 by LowPowerLab