LowPowerLab/RFM69 #114: Sketch stops/crashes if flash.initialize() is before radio.initialize()

johannstieger, 2018-12-12:
Hi, is it possible to change something on your libraries SPIFlash or RFM69 so I can initialize the flash before I initialize the radio. Now it stops on radio.initialize().

The reason for this: I wanna initialize the flash to read the UniqueId from the flash. From this ID I generate the NodeID and after that I initialize the radio with this NodeID. So I only need one Sketch/compiled binary for my 10 Moteinos.

Thanks, Johann

LowPowerLab, 2018-12-12:
I need to look into this and try to replicate the problem first, thanks for bringing this up.

sraillard, 2018-12-13:
Not related and maybe not relevant, but you can also store the NodeID in the EEPROM of the Arduino.

LowPowerLab, 2018-12-14:
Ok, I was able to replicate but only when the mote was powered from cold, without a restart. When it was restarted it was OK.
Anyway, I added SPI_transactions to the library and I tested with this sketch below, it all worked on my end, please get latest library (from the repository) and try again, I will then make a release if all is good:
```
#include <RFM69.h>    //get it here: https://www.github.com/lowpowerlab/rfm69
#include <RFM69_ATC.h>//get it here: https://github.com/lowpowerlab/RFM69
#include <SPI.h>      //included with Arduino IDE (www.arduino.cc)
#include <SPIFlash.h> //get it here: https://www.github.com/lowpowerlab/spiflash

SPIFlash flash(8, 0xEF30); //EF30 for 4mbit  Windbond chip (W25X40CL)
RFM69_ATC radio;

void setup() {
  Serial.begin(115200);
  Serial.println("START!");

  if (flash.initialize())
  {
    Serial.println("SPI Flash Init OK ... UniqueID (MAC): ");
    flash.readUniqueId();
    for (byte i=0;i<8;i++)
    {
      Serial.print(flash.UNIQUEID[i], HEX); Serial.print(' ');
    }
    Serial.println();
  }
  else
    Serial.println("SPI FlashMEM not found (is chip onboard?)");

  Serial.println("Initializing radio...");

  if (!radio.initialize(RF69_915MHZ,123,200))
    Serial.println("radio.init() FAIL");
  else
    Serial.println("radio.init() SUCCESS");

  //radio.setHighPower(); //uncomment only for RFM69HW!
  //radio.encrypt(ENCRYPTKEY);
  radio.enableAutoPower(-80);
  pinMode(LED_BUILTIN, OUTPUT);
}

byte led=0;
void loop() {
  Serial.println("Loop");
  delay(1000);
  digitalWrite(LED_BUILTIN,(led++)%2);
}
```

johannstieger, 2018-12-14:
Hi Felix, thanks for SPI_transactions in your code - my sketch now works as expected with this new library!
Thanks for your work!

closed 2018-12-14 by johannstieger