LowPowerLab/RFM69 #116: No periodic updates

NodeX-EG, 2019-01-15:
I am having this one installed and configured. When started it talks to gateway just fine, however, after random time (some time after couple of successful sends) stops sending any data.
I do not see any ACK packages nor Temp measurements sent/received.

LowPowerLab, 2019-01-15:
Just tried Gateway + Node and also RandomNumbers, everything works perfectly.
This and the forum are primarily supporting Moteino boards. I don't know how you wired your Nano.
Anyway this is not an issue. Please post any further questions in the LowPowerLab forum.
Thanks

closed 2019-01-15 by LowPowerLab

NodeX-EG, 2019-01-15:
IC2 is wiring... As said it communicates just fine, get to gateway and send
all the readings. But does it few times and stops, not sending it anything
afterwords.