LowPowerLab/RFM69 #118: Already tested all - no periodic updates

NodeX-EG, 2019-01-22:
This code simply sends few updates and then not sending anymore? How do we make sure periodic updates are sent every x minutes or something. After turned on it sends updates few times and that's all - silent from that point.

LowPowerLab, 2019-01-22:
Not an issue, please ask questions in the [forum](http://lowpowerlab.com/forum/).

closed 2019-01-22 by LowPowerLab

NodeX-EG, 2019-01-22:
this is code problem, how do you mean it is not a issue?

LowPowerLab, 2019-01-22:
It sounded like a question. There is a retry mechanism, its up to you how you use it.
If you have any code/bugs that you can demonstrate some breakage, please post it here. Otherwise questions and discussions should go in the forum, this is not a discussion board.