LowPowerLab/RFM69 #119: documentation/comprehensive examples for generic platforms missing

mamama1, 2019-02-08:
Hi,

i had quite a hard time to figure basic things out, like which pins to use on an Arduino Uno/328p or on an ESP8266. I eventually figured out that the radio constructor allows for specifiying those pins.
Also, the examples I was looking at don't really tell you much about this topic.

Many things seem not well documented (like what is the interruptNum parameter of the radio constructor doing) or at least I'm having a hard time finding the infos.

I understand that LowPowerLab focusses on its own products like Moteino, however as this is open source, I think an open issue for proper documentation on such a great project would be feasable.
Either one of the maintainers or new people coming to this library could contribute valuable stuff, that's the whole idea of OSS, afterall.

mkopa, 2019-02-08:
Everything what you must know about electrical connectors, you have got in documentations specific to your device (eg. arduino uno, esp8266 etc.)
That is data sheet of RFM69: https://cdn.sparkfun.com/datasheets/Wireless/General/RFM69HCW-V1.1.pdf

LowPowerLab, 2019-02-08:
Not an actual issue.
KLOZD

closed 2019-02-08 by LowPowerLab

mamama1, 2019-02-08:
I know how to connect the RFM now, to make it work with this library.
However, it is not obvious as @mkopa claims.

Who says, I have to use the Hardware CS pin of my particular MCU to control CS of the RFM69? Also, who says I have to use interrupt 0 of my MCU?

So no, the documentation of my MCU does not tell me, where those pins of the RFM69 have to go. This is the first library in big time I've used, which defaults to the MCU's HW-default CS pin.

I understand that you want to keep the repo's issues section clean but closing an issue which complains about the genuine lack of proper documentation also makes me understand why this library has so few open issues.

LowPowerLab, 2019-02-08:
Yes this space is for actual issues and bugs, not for feedback. We have a forum for that.
The RFM69 library was built specifically for AVR and Moteino/Arduino. It was done years ago before ESP platforms became popular. I never had an interest and need in running ESP+RFM69 together. I built it to offer a good and easy to use experience for those users, and attempt to strip away the technical advanced code that most people don't care for (ie just make this code easy to use). Advanced features and fine register control is available as well for those who want to deep dive and get to know the radio modules better (datasheets are for that reason).

Most other platforms are running fine and from what I hear and see, ESP as well for those who for whatever reason choose to run those 2 together (although there are noise and other issues with ESP which should be considered). For those reasons I did not spend time to document things that I do not care about.

If you're familiar with SPI and basic serial protocols then it should be no issues with interfacing this library or any other to any platform, regardless what assumed/default CS pin is assigned on the respective platform.
I can always add more default pins to the library to support other platforms, for that anyone can open a PR and contribute (OSS right?).