LowPowerLab/RFM69 #120: Consistent parameters for send functions

aussieW, 2019-04-01:
Hi

I have a case where I would like to send some data to some nodes which are a mix of battery power nodes which are in Listen Mode and wall powered nodes which are in RX. I notice that the function for sending data in Listen Mode takes a different parameter for the data than the normal send functions. Specifically, send() and sendWithRetry() use a const void* for buffer, whereas listenModeSendBurst() uses a void*.

i.e. 
`virtual void send(uint8_t toAddress, const void* buffer, uint8_t bufferSize, bool requestACK=false);`
`virtual bool sendWithRetry(uint8_t toAddress, const void* buffer, uint8_t bufferSize, uint8_t retries=2, uint8_t retryWaitTime=40);`
`void listenModeSendBurst(uint8_t targetNode, void* buffer, uint8_t size);`

My question is, would it be possible to make buffer a const void* in listenModeSendBurst()? That way one doesn't have to perform conversions to send the same data via these functions.

Cheers

LowPowerLab, 2019-04-04:
const void* means the buffer is not supposed to change in the scope of the function.
I concur consistency should be followed but these were contributed.
Also the ListenMode stuff has been left in the library BUT it has been concluded by a few folks that it is buggy and can get the transceiver stuck in RX under certain circumstances and drain the battery of a device prematurely. For this reason I have not really paid attention to this code much.
Have you seen the forum discussions on this and are you sure you want to use ListenMode?

LowPowerLab, 2019-05-01:
Done in [this commit](https://github.com/LowPowerLab/RFM69/commit/f072131779d311439cc84335d684c9b2b6c2c65b).

closed 2019-05-01 by LowPowerLab