LowPowerLab/RFM69 #122: Broadcast node address may be incorrectly stated

LozK, 2019-06-06:
The broadcast node address may be incorrectly stated in the comments in examples such as "Examples/Node/Node.ino" (and therefore, perhaps, other places, too).

It is correct for the previous version of this library in stating that "255 is used for broadcast", but that's no longer the case for the current version, where the node addresses are now 10-bit values, where the broadcast address has switched to being 0. 

Since that's the first I ever saw of there being such a thing as a broadcast address, it'd be nice to have it altered to match the new numbering system.

Thank you for your fine work on this incredibly useful library.

LowPowerLab, 2019-06-06:
Thanks for the wake-up call!
Folks like you keep me in check on due dilligence 👍 

I think at some point I made a mental note to go through all the examples and sources and do an overdue major cleanup and removal of lots of things including this broadcast fix and other changes.
I will leave this issue open until I manage to do all that.
I am working on other core changes which I may push first, not sure yet.

Not 100% sure what you mean by this:

> Since that's the first I ever saw of there being such a thing as a broadcast address, it'd be nice to have it altered to match the new numbering system.

Do you mean you were not familiar with what a broadcast address is? Or you meant something else?

LowPowerLab, 2019-06-06:
Ok, I pushed [this commit](https://github.com/LowPowerLab/RFM69/commit/1b2ac4a8c27d6c3b26749dbaaf68fb886d2c7b60), please check and let me know if you see any other discrepancies.

closed 2019-06-06 by LowPowerLab

LozK, 2019-06-08:
That was quick!

Sorry I didn't get back to you sooner. I'll have a check When I next get in front of my computer.

What I meant was that, when I was still quite new to this radio stuff (I suppose I still am compared to many) I came across that comment and that was the first I knew that this particular library had a broadcast node address - and if I'd come to the library recently (before you made the changes) and tried it then, it could have caused some confusion. But no worries now you've updated it! 

It might be good to have an example of how to use it properly or a comment to stress that (at least when you have more than one non-gateway node) the library user should check if a message arrived at a node using the broadcast node address and not reply to it if it did (to avoid simultaneous reply transmission from more than one node). Or to only use it for things that don't require a reply, and to make sure an ACK isn't requested by the node transmitting on the broadcast address, either.