LowPowerLab/RFM69 #124: ISR not in IRAM issue with ESP8266 2.5.2

yheynema, 2019-06-12:
I had an issue with latest ESP8266 core (2.5.2) (I'm using a Wemos D1 mini pro) where RFM69 radio init crashes the ESP with a "ISR not in IRAM" issue. This was caused by the RFM69 lib not defining the ISR handler with the fix: ICACHE_RAM_ATTR.
Here is the reference: https://github.com/esp8266/Arduino/issues/6127

LowPowerLab, 2019-06-12:
Haven't heard of such thing having to declare ICACHE_RAM_ATTR for all functions called from an ISR.
Changing all the libraries in the world to support ESP like that, seems like a ESP bug.

closed 2019-06-12 by LowPowerLab

d-a-v, 2019-06-12:
More a constraint than a bug.
Not doing so exposes the code to random crashes due to hardware requirements.

LowPowerLab, 2019-06-12:
This is specific to ESP platform correct?
I have not seen this before, so why does it happen now?
While ESP is not really an officially supported platform in this library, GPIO have been defined for it.

yheynema, 2019-06-13:
@devyte summarized here:
https://github.com/esp8266/Arduino/issues/6127#issuecomment-501493870
I implemented and tested my RFM69 project where Arduino Pro Mini talks to Wemos D1 Mini Pro. Compiled like charm, works as expected.
I tried similar to the following: https://github.com/sandeepmistry/arduino-LoRa/pull/257/files

LowPowerLab, 2019-06-13:
Thanks for the links, I see their point. They just require that attribute. I still think it's just very new and undocumented. I am not a ESP user so I cannot support the ESP platform with such specific requirements just for that platform.
The thing is ... what if every platform now requires some weird attribute to be added to each function called by an ISR or else their MCU crashes randomly, or something similar? Do we add an attribute now for each platform? My brain says "Ludicrous". They should just fix it or make it happen automatically without any attributes having to be added to every library function in the world that's called from an ISR.
Anyway I think anyone can fork the library and add those attributes.

d-a-v, 2019-06-18:
Because of *Ludicrous* I feel I can answer to that.

> I still think it's just very new and undocumented.

This [is wrong](https://github.com/esp8266/Arduino/issues/1388). The fix that is proposed to your library is indeed Ludicrous in its size or intrusion factor.
You are perfectly in the right to refuse it. That will bring more people back to your repository asking why things fail sometimes. You'll have now a good reason to close the issue: "Ludicrous" arduino core on the ESP platform.

> what if every platform now requires some weird attribute to be added

If you look at numerous arduino examples, you'll see tons of ugly `#ifdef this` and `#ifndef that`. That's the arduino way of trying to be the most compatible with every platform. The proposed change is lighter than that.

We are doing what we can to avoid that type of issue and things are not always easy as a snap of a finger.
Ram is small, all code can't go there. Ram is used as a cache for the flash, this is implemented in hardware. Cache fill needs external SPI flash transactions and that does not work well while in an ISR, so we need to direct ISR functions in ram section to avoid those random crashes. This is history on the ESP platform from the beginning.

LowPowerLab, 2019-06-18:
Why things fail ... is because ESP fails, not this library.
So while I see your point ...

Everybody knows what `#ifdef`s are. They are actually useful and necessary and been around since C existed.

In contrast, nobody knows what ICACHE_RAM_ATTR is. It's a bandaid for ESP. This library does not officially support ESP so I can't add this.

I feel the ESP/compiler should do this on the fly so we don't have to add such platform specific attributes to functions. In the real world of software this is called a `code smell`.

d-a-v, 2019-06-18:
`ICACHE_RAM_ATTR` is defined when `ESP8266` or `ESP32` is defined.
If you want to know, it's `__attribute__((section(".iram.text")))` and we use it 

> I feel the ESP/compiler should do this on the fly

... because we can't guess in source file whether a function will be called from an ISR.

>  So while I see your point ...

I don't think you do.

> This library does not officially support ESP

That should probably be stated somewhere in the main readme so you won't be bothered by people asking why it fails sometimes on their ludicrous platform.

> In the real world of software this is called a `code smell`.

I think this clearly states the end of this conversation.

LowPowerLab, 2019-06-18:
Sorry if I annoyed you.
I do wonder, did you guys open an issue on all Github repo's that have code with an ISR?
Or how do you reverse fix all the code in the world because of this obvious ESP problem that no other platform has?

devyte, 2019-06-18:
Not all Arduino-ish platforms are created equal, they all have some quirk or other that needs to be handled in some special way. A prime example is 8bit vs. 16bit vs. 32bit architecture differences. The fact that code executed while in an ISR must be in instruction RAM and not in flash is one of the quirks of the ESP/ESP32. It is related to hardware, and at this time there is no known way to work around it, or make it automatic. 
@LowPowerLab stating that we "should fix it" is not realistic, 100% compatibility across all platforms can't really be achieved when the platforms are fundamentally different. "All the code in the world" deal with such portability problems by implementing the specific quirks for the specific platforms in their library code, and this is usually done based on user demand, i. e. users request support for the platform in question. So we don't "reverse fix all the code in the world", but rather let users who require support make such requests from the library author on a case by case basis. 
You obviously have demand for support on the ESP. You can choose to support the ESP platform, in which case this quirk is needed. 
Or you can choose to not support ithe ESP (you said it isn't officially) in which case you likely should be explicit about it to your users so that they don't assume. 
Our stand is that 3rd party libraries that explicitly state support for the ESP are ok, and those that don't mention support probably contain dragons. However, new users are rarely aware of such things, and then they end up with weird crashes or unexpected behavior. We're now trying to enforce what checks we can, which resulted in a user here hitting the check in this library, which doesn't officially support the ESP, thereby stopping bad usage.

LowPowerLab, 2019-06-18:
You're basically saying the burden is on the library owners to apply the **bandaid** (it is a bandaid). I agree they have to put up with that if they support ESP.

peterfosseus, 2019-07-21:
Hi, 
Edit RFM69.cpp and change line 374 from:

void RFM69::isr0() { _haveData = true; }
to
ICACHE_RAM_ATTR void RFM69::isr0() { _haveData = true; }

Seems to fix the library?

LowPowerLab, 2019-07-25:
Done in [this commit](https://github.com/LowPowerLab/RFM69/commit/6ddfb2ff7de7f027eb7f570f1292f6f9df85bf66).