LowPowerLab/RFM69 #125: RFM69 and ESP32 freeze

spa-sam, 2019-09-28:
RFM69 and ESP32 when using the library version 1.3.0 and 1.2.0 do not pass initialization. Whereas when using the 1.1.3 library everything works fine.

LowPowerLab, 2019-09-30:
What about if you get latest head? There have been [some changes](https://github.com/LowPowerLab/RFM69/commit/6ddfb2ff7de7f027eb7f570f1292f6f9df85bf66) since 1.3.0 that address this for the ESP platform.

hermstar, 2019-10-07:
I have the same issue using a RFM69HCW module and ESP32. 

In RFM69::initialize (line 119), the system does no beyond the (first writeReg) statement:

`do writeReg(REG_SYNCVALUE1, 0xAA); while (readReg(REG_SYNCVALUE1) != 0xaa && millis()-start < timeout);`

Using https://github.com/rrobinet/RFM69X_Library I was able to successfully initialize RFM69 module (did not had time for further testing).

Any ideas?

mamama1, 2020-01-18:
I'd remove this from RFM69.cpp or at least make an #elif for the SPI_HAS_TRANSACTION condition:

 ```
SPI.setDataMode(SPI_MODE0);
  SPI.setBitOrder(MSBFIRST);
#ifdef __arm__
	SPI.setClockDivider(SPI_CLOCK_DIV16);
#else
  SPI.setClockDivider(SPI_CLOCK_DIV4); // decided to slow down from DIV2 after SPI stalling in some instances, especially visible on mega1284p when RFM69 and FLASH chip both present
#endif
```

That's redundant and appearently freezes ESP32 boards. SPI Mode, LSB/MSB and speed have been defined in the _settings object and already have been set to the SPI class via beginTransaction(_settings).

LowPowerLab, 2020-01-21:
mamama1 I just pushed [a commit](https://github.com/LowPowerLab/RFM69/commit/d8f88923776d441d5c99aa489c2ecb73992e7a11) for this.

mamama1, 2020-01-21:
wtf i have just fixed my first bug in public :D

LowPowerLab, 2020-01-21:
Did you try it with latest code?
Was it with ESP8266 or ESP32?

mamama1, 2020-01-21:
what do you mean? the change in your commit fixes the issue. i had this issue with an ESP32. when using an ESP8266, this issue doesn't occur, appearently. i was using the latest code.

LowPowerLab, 2020-01-21:
Ok, can we mark this fixed?

mamama1, 2020-01-21:
sure

closed 2020-01-21 by LowPowerLab