LowPowerLab/RFM69 #129: Question about RFM69::initialize and the reason for repeatedly retrying the write to REG_SYNVALUE1.

gitmeister, 2020-01-10:
I'm not seeing anything in the doc about this write possibly failing? Is this because it's a critical section? Are there other registers that are sensitive like this? Also, presumably, if the timeout expires initialize() should probably exit with a failure, no?

Thanks for this library.

LowPowerLab, 2020-01-13:
It has never been an issue AFAIK and seen.
Or we could assume every single register write could fail and wrap everything in fail safe returns.
If you have a demonstrable use case where such a register write fails, please open an issue here with sample code.

closed 2020-01-13 by LowPowerLab

gitmeister, 2020-01-13:
My question was misunderstood.

In this section of code in file RFM.cpp:

`
  unsigned long start = millis();
  uint8_t timeout = 50;
  **_do writeReg(REG_SYNCVALUE1, 0xAA); while (readReg(REG_SYNCVALUE1) != 0xaa && millis()-start < timeout);_**
  start = millis();
  **_do writeReg(REG_SYNCVALUE1, 0x55); while (readReg(REG_SYNCVALUE1) != 0x55 && millis()-start < timeout);_**

  for (uint8_t i = 0; CONFIG[i][0] != 255; i++)
    writeReg(CONFIG[i][0], CONFIG[i][1]);
`

there's an assumption that the write will fail and I don't see why these writes need to be retried.

LowPowerLab, 2020-01-13:
Oh that one, yes indeed.
No issues detected but that was code that was left behind since we do actually wait for the register to change value and the return time can vary.

smadurange, 2024-12-07:
Is there a reason for writing 0xAA and 0x55 to this register before doing anything else? Like the OP mentioned, there's nothing in the datasheet for the RFM69 about this.

The module doesn't work without this?

LowPowerLab, 2024-12-07:
It's a non-random preamble sequence modulated before the real transmission, and datasheet does talk about it, you just haven't looked ;-)

smadurange, 2024-12-09:
I'm using this [datasheet](https://cdn.sparkfun.com/datasheets/Wireless/General/RFM69HCW-V1.1.pdf). 

The way I understood it was that the preamble sequence (except for its length, which is programmable) was handled automatically by the RFM. I'm trying to implement the driver for learning purposes. I found it odd writing 0xAA and 0x55 to the same register in the init sequence.

Thanks for the reply. I'll check the datasheet again.

JohnOH, 2024-12-09:
I think are correct, the hardware does the framing. 

John.

LowPowerLab, 2024-12-09:
@smadurange - it's in the datasheet:
> I'm using this [datasheet](https://cdn.sparkfun.com/datasheets/Wireless/General/RFM69HCW-V1.1.pdf).
The hardware does some legwork but you have to write all bytes and tell the transceiver how to send your frames.