LowPowerLab/RFM69 #130: sometimes the network based filter fails

joysfera, 2020-01-26:
*Sometimes* a gateway receives packet sent by a node with a different networkID.

It's rather rare (like once in a week) but it does happen.

Encryption is enabled and the encryption keys are the same in both networks. Gateway and Node IDs are the same as well.

I thought it was impossible to receive packet for a different network. What can I do to truly isolate the networks so that the packets for different networks cannot be received?

LowPowerLab, 2020-01-28:
Since there's no variable to see the network ID (it is indeed stored in and filtered by second SYNC register), how do you verify or log that that stray packet is from another network?

joysfera, 2020-01-28:
say one node in network X transmits temperature, while the other node in network Y transmits humidity. If I receive humidity in network X I know the packet came from node in network Y.

LowPowerLab, 2020-01-28:
Ah I can see how this can happen. You said everything is the same except the network ID. The sync word part of the message is not encrypted. If that network byte is corrupted during transmission, and the values are even slightly similar (in bits) it's possible the corrupted flipped bit(s) to rarely pass this filter for another network ID. I would either:
- add more SYNC bytes (up to 8)
- make the network IDs bit patterns complementary (to decrease the chance that 1 or a few flipped bits will match another valid network ID)
- change the encryption keys
So closing this now since it does not sound like this is a library bug.

closed 2020-01-28 by LowPowerLab

joysfera, 2020-01-28:
Thanks. The network ID could be repeated several times at the 6 remaining empty slots, that's a good idea.