LowPowerLab/RFM69 #131: Minor changes to eliminate compiler warnings.

MojaveTom, 2020-01-28:
RFM69.cpp line 331 is an unused variable.
RFM69_ATC.cpp line 129 is an unused variable.
RFM69_OTA.cpp line 508 should use editing code "%hu" to convert a uint16_t.

LowPowerLab, 2020-01-28:
Thanks, What's the warning you get for using %u ?
I see no functional difference.

MojaveTom, 2020-01-28:
Regarding the compiler warning in RFM69_OTA.cpp

Using VSCode/PlatformIO.  Compiler is: arm-none-eabi-g++ 

Produces this warning with original code:

RFM69_ID92/RFM69_OTA.cpp:508:58: warning: format '%u' expects argument of type 'unsigned int*', but argument 3 has type 'uint16_t* {aka short unsigned int*}' [-Wformat=]
         sscanf((const char*)radio.DATA, "FLX:%u:OK", &tmp);

Scan field code %hu is explicitly for reading uint16_t.

—————————————————————

On a different topic.  It costs nothing to put:

#include <Arduino.h>

at the top of each main program file, and makes porting to VSCode simpler.  (I don’t have to add that line when I import the .ino file as a .cpp.)

Also, Arduino IDE scans your .ino file and creates function signatures for you.  It would be nice if you would explicitly put them in.

It’s too bad that the PlatformIO .ino import function doesn’t do both of those things for us.

—————————————————————— 

I’m still interested in buying the current version MightyHat when they become available.  Knowing a new version is “in the works” means I would buy fewer at this time.  Let’s see:
HomeAssistant Pi, MoteinoHub Pi  in Steamboat
HomeAssistant Pi, MoteinoHub Pi, remote control Pi in Ridgecrest

That’s at least 5 Pis that I would like to protect “soon”.  When the new revision comes out, I would move those MightyHats to some less critical uses (PiHole, Octoprint, alternate MoteinoHubs…) and get new ones for the critical systems.

Thanks,
Tom

LowPowerLab, 2020-01-28:
Interesting so it complains about that instance but not the many others in the file?
-------------------------------------
For the other things, are you talking about the example sketches in the library?
There are many of those, it actually does cost some time and maintenance thereafter whenever the winds of change blow and it seems to happen more often that I wish for.
I can certainly appreciate others using better IDEs. However I could not support all these other platforms.
I would certainly add it if it was required by Arduino (my main support platform). I believe any other platforms that import arduino files (which the .ino file extension designates) should in turn implement this behind the scenes in their IDEs. So IMHO this issue should be raised with them instead of requiring this for * arduino sketches in the world.

LowPowerLab, 2020-04-07:
Fixed in  dc5d2f8 and aec150f.

closed 2020-04-07 by LowPowerLab