LowPowerLab/RFM69 #136: ESP32 SPI2 & SPI3 Compatible Code (Cannot Use Library Without it)

ajaybnl, 2020-03-30:
I have sucessfully tested the ESP32 connected to RFM69HCW with this library by adding these lines:

in RFM69.cpp Near SPI.begin(); (Replace)
```

 #if defined(ESP32)
//Choose One SPI PORT
  SPI.begin(18,19,23,5);   //SPI3  (SCK,MISO,MOSI,CS)
// SPI.begin(14,12,13,15);   //SPI2   (SCK,MISO,MOSI,CS) 
  #else
SPI.begin();
  #endif
```

and in RFM69.h: 
```

#define RF69_IRQ_PIN    2
#define RF69_SPI_CS     5
```

Otherwise the Library does not work.

Please add these lines for ESP32.

LowPowerLab, 2020-03-30:
@ajaybnl 
These are specific pins to some specific ESP32 variant.
Which ESP32 board are these pins from?

ajaybnl, 2020-03-30:
> @ajaybnl
> These are specific pins to some specific ESP32 variant.
> Which ESP32 board are these pins from?

Esp32 Wroom Module (Without psram)

ajaybnl, 2020-03-30:
[image]

LowPowerLab, 2020-03-30:
I am talking about ARDUINO BOARD DEFINITION. Which one is that? Not the module itself.........................

ajaybnl, 2020-03-30:
> I am talking about ARDUINO BOARD DEFINITION. Which one is that? Not the module itself.........................

Im uploading using esp-dev-kit variant.

ajaybnl, 2020-03-30:
Im making rfm69 and espnow gateway ( code: https://github.com/ajaybnl/esp32_rfm69_espnow_mqtt_gateway)

ajaybnl, 2020-03-30:
Or you can impliment spiClass selection like this library: https://github.com/JF002/cfRFM69/tree/master/src

LowPowerLab, 2020-03-31:
I understand now, OK i have merged this in the library. Done in commit 1e1236d4b4f3054d130249e6d4a2ea0a14ea98c1
Thanks

closed 2020-03-31 by LowPowerLab

ajaybnl, 2020-03-31:
Cheers.