LowPowerLab/RFM69 #138: Missing Serial.println in "RFM69_OTA.cpp"

Spexx-, 2020-04-06:
Hi Felix,

until recently, I have used the older "WirelessHEX69" library for wireless programming of my Moteinos. For some days ago I have updated the library to the current version (1.4.0) and I have found out that there is a slight difference between the both libraries, which will cause a problem with my selfmade windows application for wireless programming.

I will check the end of line from the WirelessProgramming-Gateway via a carriage return and newline charakter, but this is missing in the "PrintHex83" subfunction:

**Old library, "WirelessHEX69.cpp":**
```
void PrintHex83(uint8_t *data, uint8_t length) // prints 8-bit data in hex
{
  char tmp[length*2+1];
  byte first ;
  int j=0;
  for (uint8_t i=0; i<length; i++)
  {
    first = (data[i] >> 4) | 48;
    if (first > 57) tmp[j] = first + (byte)39;
    else tmp[j] = first;
    j++;
    first = (data[i] & 0x0F) | 48;
    if (first > 57) tmp[j] = first + (byte)39;
    else tmp[j] = first;
    j++;
  }
  tmp[length*2] = 0;
  Serial.println(tmp);
}
```

**New library, "RFM69_OTA.cpp":**
```
void PrintHex83(uint8_t* data, uint8_t length) 
{
  uint8_t tmp;
  for (uint8_t i=0; i<length; i++) 
  {
    tmp = (data[i] >> 4) | 48;
    Serial.print((char)((tmp > 57) ? tmp+7 : tmp));	
    tmp = (data[i] & 0x0F) | 48;
    Serial.print((char)((tmp > 57) ? tmp+7 : tmp));
  }
}
```

Can you please add a Serial.println() to the end of the PrintHex83 subfuction so that there will be the identical behavior compared to the old version of the library?

LowPowerLab, 2020-04-07:
@Spexx- Thanks a lot, very good fix!

closed 2020-04-07 by LowPowerLab

Spexx-, 2020-04-07:
Wow thank you, that was a very fast fix :)

I'm already using the Moteinos for several years and I still love them. Thank you very much for your great work and sharing all these information!

commit 24d63d8f (2020-04-07): Fix #138