LowPowerLab/RFM69 #140: [Question] NRF5 Project

klerone, 2020-05-01:
Hello Felix, first thank you for the awesome job you have done with this library. 

I am hobbyist interested to try the NRF5 platform, I am in China and here this boards are really cheap and easy to get. 

I have read the code and found reference to this chips in the RFM69.h file, also I have read in your blog but I cannot find if someone else have done some experiments with it. Have you try it, what is your experience?. 

Thank you in advance.

Best regards.

Francisco.

LowPowerLab, 2020-05-01:
Sorry but I can't help with that. I tried NRF and concluded it is :poop:

closed 2020-05-01 by LowPowerLab

klerone, 2020-05-01:
well, that is a great visual description. Thank you Felix!