LowPowerLab/RFM69 #146: Not working for Tiny anymore.

mpruessmeier, 2020-06-29:
After including LowPower.h with Commit 64eb36ba852be43f56bb555c1d7d5a0ac52e0873 from May 18. 
code is not working with ATTiny any more cause LowPower.h is`nt supporting them.

Versions before May 18. are working well.

amadeuspzs, 2020-09-03:
Yes, I have been trying to get this working on an ATTiny and struggling.

For now I am getting things working with debug output on a 32U4 (Sparfun Arduino Pro Micro clone) and will attempt to get it working on an ATTiny.

There are several issues with ATTiny around memory size, but also software serial availability for debugging as well as flipped MISO/MOSI pins (the pinout of the ATTiny is designed for it to be a slave during ICSP as opposed to a controller).

LowPowerLab, 2020-09-03:
So sorry but Attiny is not a suitable platform for this library and is not officially supported.
Maybe if you submit a PR with proper defines I can merge it. But it really depends because i dont want to pollute the library with #ifdefs for non supported platforms.

closed 2020-09-03 by LowPowerLab