LowPowerLab/RFM69 #148: SPI_CLOCK_DIV2 - too fast for STM32

janhrach, 2020-07-26:
Hi,

I have been using 1.4.0 version with STM32F103C8T6 (`__STM32F1__` https://github.com/rogerclarkmelbourne/Arduino_STM32 ) with no issues.

1.4.1 version changes SPI_CLOCK_DIV16 to DIV2 for all non `__SAMD51__` devices. With DIV2, however, STM32F103 stopped working (stalls).

Would it be possible to add support for STM32 back? Or better for all non `__SAMD51__` arm devices in general?

This works for me:

```
$ diff -u RFM69.cpp_orig RFM69.cpp
--- RFM69.cpp_orig	2020-07-08 15:28:42.000000000 +0200
+++ RFM69.cpp	2020-07-26 23:17:05.188984100 +0200
@@ -498,7 +498,9 @@
   // set RFM69 SPI settings explicitly
   _spi->setDataMode(SPI_MODE0);
   _spi->setBitOrder(MSBFIRST);
-  #ifdef defined(__SAMD51__)
+  #if defined(__SAMD51__)
+    _spi->setClockDivider(SPI_CLOCK_DIV16);
+  #elif defined(__arm__)
     _spi->setClockDivider(SPI_CLOCK_DIV16);
   #else
     _spi->setClockDivider(SPI_CLOCK_DIV2);
```

Thanks,
Peter

LowPowerLab, 2020-07-29:
What is the F_CPU for your STM32 device?
Doesn't that architecture support SPI_TRANSACTIONS? It should.
In which case the SPI clock speed should be set to 8mhz and the code should execute this and not the explicit SPI clock divider:

```
#ifdef SPI_HAS_TRANSACTION
  _settings = SPISettings(8000000, MSBFIRST, SPI_MODE0);
#endif
```

janhrach, 2020-07-30:
F_CPU for this device can be either one of: 48000000, 72000000 (default), 128000000

Looking at the SPI library on that specific arduino STM32 project, SPI_HAS_TRANSACTION is not defined:

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/a4007f57dfaf26a0e92191a261e312888b896995/STM32F1/libraries/SPI/src/SPI.h#L49

Confirmed also by trying to compile a sketch that would print SPI_HAS_TRANSACTION. Compilation fails with:
'SPI_HAS_TRANSACTION' was not declared in this scope

LowPowerLab, 2020-07-30:
@janhrach No SPI_HAS_TRANSACTION - that's a bad thing.
Anyway, what's the best most generic directive that can capture all the STM32 devices? __STM32__?

Or I can just do something like:
```
#if defined(__arm__)
  _spi->setClockDivider(SPI_CLOCK_DIV16);
#else
   _spi->setClockDivider(SPI_CLOCK_DIV2);
#endif
```

closed 2020-08-01 by LowPowerLab

janhrach, 2020-08-03:
I tried several DIVs in combination with various CPU speeds, and also having an tft display as a second device connected to either SPI1 or SPI2 of STM32F103.

The outcome is that the best divider is SPI_CLOCK_DIV8. It is the lowest divider (fastest speed) using which both the RFM69 device and the tft display work at 72Mhz and 128Mhz CPU speed.

I do not know what is the most generic directive that can capture all the STM32. I don't have other STM32 devices than STM32F103 (directive `__STM32F103__` in the core I use). For now, I would use `__arm__`. It may cover other devices that would fall under this directive and may need DIV8 too. A better directive can be added later, if found, to narrow down the devices. The "official" ST's STM32 core, which I cannot use for my project, has the SPI_HAS_TRANSACTION defined anyway.

If you would like to keep DIV16 for `__SAMD51__`, then please, apply this diff:

```
$ diff -u RFM69.cpp_orig RFM69.cpp
--- RFM69.cpp_orig      2020-07-08 15:28:42.000000000 +0200
+++ RFM69.cpp   2020-08-03 12:48:23.615673900 +0200
@@ -498,8 +498,10 @@
   // set RFM69 SPI settings explicitly
   _spi->setDataMode(SPI_MODE0);
   _spi->setBitOrder(MSBFIRST);
-  #ifdef defined(__SAMD51__)
+  #if defined(__SAMD51__)
     _spi->setClockDivider(SPI_CLOCK_DIV16);
+  #elif defined(__arm__)
+    _spi->setClockDivider(SPI_CLOCK_DIV8);
   #else
     _spi->setClockDivider(SPI_CLOCK_DIV2);
   #endif
```

Thanks,
P

LowPowerLab, 2020-08-03:
Sorry, I added __arm__ for div16 and div2 for anything else. There are __arm__ boards that run at 120, 180mhz.
This code branch is really archaic and should not be used. Or use the official core with SPI_TRANSACTION support. Or if thats impossible you can make a local code change specific to your project.
Please do a proper pull which I can merge.

janhrach, 2020-08-04:
Created pull request https://github.com/LowPowerLab/RFM69/pull/149 specifically for STM32F103 board only.

Please be kind to me if I did not do it properly, it's my very first pull request.

Thanks,
P

LowPowerLab, 2020-08-04:
Merged, thanks!