LowPowerLab/RFM69 #153: New dependency introduced in version 1.4.1

sraillard, 2020-09-20:
Before version 1.4.1 of the library, there wasn't a dependency on **LowPower** library (http://github.com/LowPowerLab/LowPower).

This seems to be introduced in this commit: https://github.com/LowPowerLab/RFM69/commit/64eb36ba852be43f56bb555c1d7d5a0ac52e0873

In Arduino IDE, building a sketch is failing with the version 1.4.1 as this library wasn't installed before.
Can it be optionnal?

LowPowerLab, 2020-09-22:
True. It helps achieve a low and more granular sleeping mode than with the WDT (using "Listen Mode").
I would like to support this on AVR (328p and 1284p) and SAMD (21, 51) chips.
How do you propose this is implemented, since those sleep functions are in the LowPower library?
https://github.com/LowPowerLab/LowPower

sraillard, 2020-09-25:
Using your library (and a lot of Moteino) for few years, I have seen a lot of work and added features.

Maybe using a couple of #define to enable/disable some extra features like ATC, OTA or Listen Mode would allow users to only keep what they need with their dependencies? Listen Mode depends on LowPower but OTA depends on SPIFlash. I'm personally using the OTA feature but I understand user not using it to be able to disable it and remove the SPIFlash dependency.

Maybe also explaining the dependencies needed in the README.md would be good.

LowPowerLab, 2020-09-29:
Fixed in e9f527e, there is now also a new release to include this - [1.4.2](https://github.com/LowPowerLab/RFM69/releases/tag/1.4.2).

closed 2020-09-29 by LowPowerLab

sraillard, 2020-09-29:
Can you update the library.json and library.properties files with the new version 1.4.2 so Arduino IDE can find it?

LowPowerLab, 2020-09-29:
Just did and re-released, please check. Thanks

sraillard, 2020-09-30:
Working fine, perfect!