LowPowerLab/RFM69 #161: 1284P and RFM69 hangs

wvdakker, 2021-02-13:
Hi 
I have a RFM69 attached to a 1284p.
fuses; lfuse:0xc2 hfuse:0xd1 efuse:0xff:

Attached the pins (using the bobuino pinout)
MOSI: PB5 
MISO: PB6
SCK :  PB7
SS : PD7
INTR PIN: PD2 / INTR NUM: 0

When sending a package the RFM driver is hanging in the SendFrame function
" while ((readReg(REG_IRQFLAGS2) & RF_IRQFLAGS2_PACKETSENT) == 0x00); // wait for PacketSent"
Where the output of readReg(REG_IRQFLAGS2)  keeps at 48;

Initializing the driver gives no problem._interrupPin, _interruptNum and _slaveSelectPin are initialized
with the proper values (checked with readback).

Missed I something essential or is it a bug?

wvdakker, 2021-02-14:
Problem solved. Bobuino vs Standardpin mix-up

closed 2021-02-14 by wvdakker

martin323232, 2021-07-31:
@wvdakker How did you solve this issue?
I'm using INT2, that is D6 on Bobuino

wvdakker, 2021-08-01:
No not on 20Mhz, it is working on 8Mhz internal.
For now it is ok for me.

martin323232, 2021-08-04:
Got it working on INT2, was an issue with defective RFM69 unit.
No need to edit IRQ pin in library