LowPowerLab/RFM69 #163: RFM69 on ESP32

carlkesselman, 2021-02-22:
Hi,

I have found two issues with the RFM69 library when running on an ESP32 using the Arduino framework.  They are:

1) The initialization for the SPI interface is not correct, at least for the feather.  Just using 
`  _spi->begin();`
 works fine.  Using the currently configured: 
`_spi->begin(18,19,23,5); `
does not.
2) You need to set the pinMode on the interrupt for it to function properly. Adding the line: 
`pinMode(_interruptPin, INPUT); `
seems to solve the problem.

Thanks,

Carl

LowPowerLab, 2021-02-22:
Hi @carlkesselman 
I don't use ESP32 with RFM69. I merged some changes suggested by various people. Probably not a great thing since I can't properly test. Then there's a gazillion variants of ESP32 boards each with various pinouts.
Here's an issue that calls for the specific pins, to make things work:
https://github.com/LowPowerLab/RFM69/issues/136

So how will this change affect that and other ESP32 variants?

klerone, 2021-05-16:
Hi @carlkesselman 

I am currently trying to make a gateway with the ESP32 but I cannot get it to work.

I have the wiring like this:
**ESP32**             **RFM69**
GPIO5            SS
GPIO18          SCK
GPIO19          MISO
GPIO23          MOSI
GPIO2            DIO0       

RFM69 is powered correctly with an AMS1117 3.3V, the ESP32 is powered from the USB, both connected by the GND, the sender is working correctly (I have an atmega328 to test it is sending correctly) 

I already tried both of your suggestions but still don't work, hope you can point me to a solution.

thanks.

Fran Calero.

mrSilkie, 2021-08-29:
Hi, I have also had trouble getting this to work on esp32. I am using a lora pi board and trying to make it work. I also doubt that the RFM69 requires 5v as its a 3v3 board

I turn the onboard LED on at the start of setup and off at the end of setup() and the light never turns off. Running the TxRxBlinky example.

It seems that when it get to the radio.send it hangs. Im not sure if this is normal behaviour but I am rather stuck. I've spent some time looking for another library but I think this one is by far the best and has sucked away a lot of the development time from other projects. I also understand that this was made before the esp32 was even a thing so much respect.

I suggest a simple

` radio.set_pins(MISO_PIN,` MOSI_PIN . . . ); 

could be used to help people with non standard setups to use this library. I'm still stuck trying to pinpoint where the issue is, is it hardware, is it software, is the wiring incorrect. I'm not sure. I really want to try the most simple library I can as I also had issues running the SPIFlash.h library used in many of the examples.

LowPowerLab, 2021-09-01:
@carlkesselman, @mrSilkie:
The library should not have to so anything like this (currently in initialize for ESP boards, added at some point because it worked for someone but it's a total code smell): `_spi->begin(18,19,23,5);`

Instead, the _spi instance (if different than the default) can be setup and passed into the constructor, this is more proper.
For instance for a SAMD board that would be something like this:
`SPIClass mySPI(&sercom1, 12, 13, 11, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_3);`

Perhaps the _interruptPin needs to be passed as well if it's not the default. There is a [section in the header file](https://github.com/LowPowerLab/RFM69/blob/master/RFM69.h#L139) where the default is set to 4/2 respectively for ESP8266/ESP32. Anyway there is also the option to pass the `irqPin` to the constructor and then maybe what is missing is the library to call `pinMode(irqPin, INPUT)`. Someone needs to test this properly and submit a PR.

DerLang1975, 2022-10-05:
Hello,

I do not get it to work with ESP32 DEVKITC as well.
I've debugged it to the point in sendFrame(..)

while ((readReg(REG_IRQFLAGS1) & RF_IRQFLAGS1_MODEREADY) == 0x00); // wait for ModeReady

Here it hangs forever. I've checked the setMode(RF69_MODE_STANDBY);
Having
Serial.println(readReg(REG_OPMODE), HEX);
  Serial.println(readReg(REG_IRQFLAGS1), HEX);
before the while loop prints:
4
0
=> hence correct mode set, but irq flags not updated... 

ReadReg in Setup:

Address - HEX - BIN
1 - 4 - 100
2 - 0 - 0
3 - 1A - 11010
4 - 40 - 1000000
5 - 3 - 11
6 - 52 - 1010010
7 - D9 - 11011001
8 - 0 - 0
9 - 0 - 0
A - 1 - 1
B - 0 - 0
C - 2 - 10
D - 92 - 10010010
E - F5 - 11110101
F - 20 - 100000
10 - 24 - 100100
11 - 7F - 1111111
12 - 9 - 1001
13 - F - 1111
14 - 40 - 1000000
15 - B0 - 10110000
16 - 7B - 1111011
17 - 9B - 10011011
18 - 8 - 1000
19 - 86 - 10000110
1A - 8A - 10001010
1B - 40 - 1000000
1C - 80 - 10000000
1D - 6 - 110
1E - 10 - 10000
1F - 0 - 0
20 - 0 - 0
21 - 0 - 0
22 - 0 - 0
23 - 2 - 10
24 - FF - 11111111
25 - 40 - 1000000
26 - 7 - 111
27 - 0 - 0
28 - 0 - 0
29 - FF - 11111111
2A - 0 - 0
2B - 0 - 0
2C - 0 - 0
2D - 3 - 11
2E - 98 - 10011000
2F - 55 - 1010101
30 - 64 - 1100100
31 - 0 - 0
32 - 0 - 0
33 - 0 - 0
34 - 0 - 0
35 - 0 - 0
36 - 0 - 0
37 - 10 - 10000
38 - 40 - 1000000
39 - 0 - 0
3A - 0 - 0
3B - 0 - 0
3C - 8F - 10001111
3D - 11 - 10001
3E - 73 - 1110011
3F - 61 - 1100001
40 - 6D - 1101101
41 - 70 - 1110000
42 - 6C - 1101100
43 - 65 - 1100101
44 - 45 - 1000101
45 - 6E - 1101110
46 - 63 - 1100011
47 - 72 - 1110010
48 - 79 - 1111001
49 - 70 - 1110000
4A - 74 - 1110100
4B - 4B - 1001011
4C - 65 - 1100101
4D - 79 - 1111001
4E - 1 - 1
4F - 0 - 0

DerLang1975, 2022-10-05:
I've checked further more. Registers are nearly all what I expected, so SPI seems to work as well as setting the values. (i.e. Network ID was correctly set).

But still, the IRQ Registers are not set, or they are cleared before they could have been read by the code.

Any idea why IRQ registers are not set or what could cause this strange behaviour?

haschler, 2022-11-18:
Do you by chance use a RFM69(H)CW? If so, try connecting RST of the radio module to GND...

haschler, 2022-11-18:
Btw I think this issue can be closed. Perhaps it would be useful to have an example for the ESP with wiring instructions.

LowPowerLab, 2022-11-20:
Should be fixed by https://github.com/LowPowerLab/RFM69/pull/185

closed 2022-11-20 by LowPowerLab