LowPowerLab/RFM69 #164: New release for Arduino IDE

sraillard, 2021-05-03:
The last release that can be included from the libray manager in the Arduino IDE is the version 1.4.2 from last September.
I have seen that there are some new modifications and the current version 1.4.2 is compiled with warnings.
Do you plan to make a new release?

LowPowerLab, 2021-05-03:
@sraillard What board/version are you compiling for? And which IDE version?

sraillard, 2021-05-03:
I'm using the latest Arduino IDE (version 1.8.13 for Windows) and my boards are the "classic" Moteino.

LowPowerLab, 2021-05-03:
OK thanks, have you verified with the latest changes it does not compile with any warnings?

sraillard, 2021-05-03:
I'll do a quick review of the master code, and keep you updated

LowPowerLab, 2021-05-03:
Ok, the reason I asked is because I can see no warnings with 1.4.2 on IDE 1.8.13 or older versions.
So I wonder where the discrepancy is.
Anyway I could make a release but I'd like to find out why you're getting warnings. Maybe can you include some of those warnings here, and perhaps use a generic sketch like one of the unmodified examples in the library (Node).

sraillard, 2021-05-03:
Ok, I have done a quick test with the following setup:
* Arduino IDE 1.8.13 (Windows 10)
* Arduino AVR Boards v1.8.3
* Moteino AVR Boards v1.6.1
* RFM69 v1.4.2
* SPI Flash 101.1.3

I have opened the example "node.ino" and compiled it, I have the follwing warnings:

```
In file included from C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\Examples\Node\Node.ino:27:0:
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab/RFM69.h: In constructor 'RFM69::RFM69(uint8_t, uint8_t, bool, uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab/RFM69.h:199:81: warning: unused parameter 'interruptNum' [-Wunused-parameter]
     RFM69(uint8_t slaveSelectPin, uint8_t interruptPin, bool isRFM69HW, uint8_t interruptNum) //interruptNum is now deprecated
                                                                                 ^~~~~~~~~~~~
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab/RFM69.h: In member function 'virtual void RFM69::interruptHook(uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab/RFM69.h:241:40: warning: unused parameter 'CTLbyte' [-Wunused-parameter]
     virtual void interruptHook(uint8_t CTLbyte) {};
                                        ^~~~~~~
```

and

```
In file included from C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.cpp:26:0:
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h: In constructor 'RFM69::RFM69(uint8_t, uint8_t, bool, uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h:199:81: warning: unused parameter 'interruptNum' [-Wunused-parameter]
     RFM69(uint8_t slaveSelectPin, uint8_t interruptPin, bool isRFM69HW, uint8_t interruptNum) //interruptNum is now deprecated
                                                                                 ^~~~~~~~~~~~
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h: In member function 'virtual void RFM69::interruptHook(uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h:241:40: warning: unused parameter 'CTLbyte' [-Wunused-parameter]
     virtual void interruptHook(uint8_t CTLbyte) {};
                                        ^~~~~~~
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.cpp: In member function 'bool RFM69::initialize(uint8_t, uint16_t, uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.cpp:62:21: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (_interruptNum == NOT_AN_INTERRUPT) return false;
In file included from C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69_ATC.h:31:0,
                 from C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69_ATC.cpp:29:
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h: In constructor 'RFM69::RFM69(uint8_t, uint8_t, bool, uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h:199:81: warning: unused parameter 'interruptNum' [-Wunused-parameter]
     RFM69(uint8_t slaveSelectPin, uint8_t interruptPin, bool isRFM69HW, uint8_t interruptNum) //interruptNum is now deprecated
                                                                                 ^~~~~~~~~~~~
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h: In member function 'virtual void RFM69::interruptHook(uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h:241:40: warning: unused parameter 'CTLbyte' [-Wunused-parameter]
     virtual void interruptHook(uint8_t CTLbyte) {};
                                        ^~~~~~~
In file included from C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69_OTA.h:37:0,
                 from C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69_OTA.cpp:33:
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h: In constructor 'RFM69::RFM69(uint8_t, uint8_t, bool, uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h:199:81: warning: unused parameter 'interruptNum' [-Wunused-parameter]
     RFM69(uint8_t slaveSelectPin, uint8_t interruptPin, bool isRFM69HW, uint8_t interruptNum) //interruptNum is now deprecated
                                                                                 ^~~~~~~~~~~~
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h: In member function 'virtual void RFM69::interruptHook(uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69.h:241:40: warning: unused parameter 'CTLbyte' [-Wunused-parameter]
     virtual void interruptHook(uint8_t CTLbyte) {};
                                        ^~~~~~~
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69_OTA.cpp: In function 'uint8_t sendHEXPacket(RFM69&, uint16_t, uint8_t*, uint8_t, uint16_t, uint16_t, uint16_t, uint8_t)':
C:\Users\Sebastien\Documents\Arduino\libraries\RFM69_LowPowerLab\RFM69_OTA.cpp:512:59: warning: format '%hu' expects argument of type 'short unsigned int*', but argument 3 has type 'uint16_t* {aka unsigned int*}' [-Wformat=]
         sscanf((const char*)radio.DATA, "FLX:%hu:OK", &tmp);
                                                       ~~~~^
```

sraillard, 2021-05-03:
I have tested with the latest "master" version, and I have the same warnings.

LowPowerLab, 2021-05-03:
Weird, I have the same versions of everything, tested both the master/head and 1.4.2, no warnings.
The warnings seem bogus though, for instance CTLbyte is used in RFM69_ATC.h in a function override in a polymorphic fashion, and that should be evident to the compiler since RFM69_ATC.h is included in the sketch.

sraillard, 2021-05-03:
I think there are some valid warnings, I'm preparing a PR.

Meanwhile, here are my setttings, maybe the **All** isn't the default setting?

[image]

sraillard, 2021-05-03:
Here is the PR : https://github.com/LowPowerLab/RFM69/pull/165

LowPowerLab, 2021-05-03:
Fixed in Pull #165

closed 2021-05-03 by LowPowerLab

sraillard, 2021-05-03:
Can you make a new release for the Arduino IDE?

LowPowerLab, 2021-05-03:
Sure, working on a few things and I will have this ready asap.

sraillard, 2021-05-04:
Perfect, thank you!