LowPowerLab/RFM69 #176: Incorrect power level at RFM69 class definition for Highpower optioj

rrobinet, 2022-03-10:
While defining a RFM69 class including a high-power transceiver the maximum power level for high-power transceivers is not correctly updated as from the new power levels settings (always default i31)
Typically
#define slaveSelectPin 	2
#define interruptPin 	10
#define isRFM69HW	true
#define isRFM69HW_HCW	true
RFM69 radio (slaveSelectPin, interruptPin, isRFM69HW);
or 
 RFM69 radio (slaveSelectPin, interruptPin, isRFM69HW_HCW, &SPI);
Gives via radio.getPowerLevel : 31 while its should be following the new power settings 23
**Suggestion**

Modify the class as follow:
RFM69::RFM69(uint8_t slaveSelectPin, uint8_t interruptPin, bool isRFM69HW_HCW, >
  _slaveSelectPin = slaveSelectPin;
  _interruptPin = interruptPin;
  _mode = RF69_MODE_STANDBY;
  _spyMode = false;
  _isRFM69HW = isRFM69HW_HCW;
  **if (_isRFM69HW) _powerLevel = 23;
  else _powerLevel = 31;**
  _spi = spi;
#if defined(RF69_LISTENMODE_ENABLE)
  _isHighSpeed = true;
  _haveEncryptKey = false;
  uint32_t rxDuration = DEFAULT_LISTEN_RX_US;
  uint32_t idleDuration = DEFAULT_LISTEN_IDLE_US;
  listenModeSetDurations(rxDuration, idleDuration);
#endif
}

LowPowerLab, 2022-03-10:
I believe there is no issue, if you use the library **correctly**. The constructor does not initialize a lot of critical registers, so assuming that just calling the constructor will get everything ready is incorrect. Just take a look at what initialize() does, to see what I mean, without it, nothing works regardless  if we adjust power in the constructor. We would have to move initialize() code entirely in the constructor to make things work.

The call to radio.initialize() is **absolutely required** after the constructor. That will call setPowerLevel() which adjusts everything according to the isHW_HCW parameter - it calls setHighPower().

If you DO NOT call initialize() the power level is left at 31 which is correct (for CW, W modules), regardless what you call in the constructor. But this is irrelevant, we can only discuss what happens when the library is used as intended, when initialize() is called before any other usage of the radio module.

closed 2022-03-10 by LowPowerLab

rrobinet, 2022-03-10:
OK
I was little bit too fast on that one, but it is a little bit tricky.
The trick resides in the setHighPower function called by the init which enforce a power level of 23 if greater that 23 (which is the case) ; the _powerLevel variable in never updated (if no explicit setPowerLevel of course )
Sorry about that problem report

LowPowerLab, 2022-03-10:
Actually _powerLevel  is set by setPowerLevel(), which is called by setHighPower() in initialize() regardless of what module is used.
Hence the obvious, and unavoidable need to call initialize(), not even sure how you could use the library without it, unless you set registers yourself, in which case you wouldn't need the library ;-)

rrobinet, 2022-03-10:
and sorry again, the _powerLevel is well adapted by the setPowerLevel function called by the setHighPower

LowPowerLab, 2022-03-10:
No problem! 👍