LowPowerLab/RFM69 #177: Setup with constructor

charles-d-burton, 2022-03-13:
So, I'm trying to setup the radio via the constructor and I'm never getting anything back from the serial console.  I'm setting it up with the following sketch.  I was wondering if there's a flag I can set somewhere to get debugging information.
```cpp
// LowPowerLabs RFM69 Library
#include <RFM69.h>
// SPI Library
#include <SPI.h>
// Addresses for this node. CHANGE THESE FOR EACH NODE!
volatile byte state = LOW;

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // My node ID
#define TONODEID      1   // Destination node ID

#define FREQUENCY     RF69_915MHZ

// AES encryption (or not):

#define ENCRYPT       true // Set to "true" to use encryption
#define ENCRYPTKEY    "passwordpassword" // Use the same 16-byte key on all nodes

// Use ACKnowledge when sending messages (or not):
#define USEACK        true // Request ACKs or not

#define CSPIN         17
#define IRQPIN        20

// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,&SPI);

void setup() {
  SPI.setRX(16);
  SPI.setTX(19);
  SPI.setCS(17);
  SPI.setSCK(18);  
  SPI.begin(); 
  
  pinMode(IRQPIN, INPUT);
   // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.print("Node ");
  Serial.print(MYNODEID,DEC);
  Serial.println(" ready");
  // Initialize the RFM69HCW:
  radio.initialize(FREQUENCY, MYNODEID, NETWORKID);
  radio.setHighPower(); // Always use this for RFM69HCW

  // Turn on encryption if desired:

  if (ENCRYPT) {
    radio.encrypt(ENCRYPTKEY);
  }

  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial)delay(10);
  delay(1000);
}

void loop() {
  Serial.println(radio.canSend());
  Serial.print("RSSI: ");
  Serial.println(radio.readRSSI());
  Serial.print("Temp: ");
  Serial.println(radio.readTemperature());
  digitalWrite(LED_BUILTIN, state);
  blink();
  delay(500);
}

void blink() {
  state = !state;
}
```

```
0
RSSI: 0
Temp: 165
```

rrobinet, 2022-03-13:
Fortunately I am playing with the last RFM69 library with a Moteino. So I did had the opportunity to test your code.
I don't know what you are trying to achieve and on which kind of board, but you should verify your SPI and SPI library configuration and the SS ad IRQ of the RFM transceiver. 
A simple test is to check the state of the Initialize function which is a boolean one, if this one fails no need to expect  any results.
Settings for a Moteino (ATmega328P) works perfectly. 
See modified code below:
 * SPI functions removed
 * CSPIN         10 // For Moteino
 * IRQPIN        2  // For Moteino
*  Serial.begin at beginning of setup
* RFM69 initialization check added

```
// LowPowerLabs RFM69 Library
#include <RFM69.h>
// SPI Library
#include <SPI.h>
// Addresses for this node. CHANGE THESE FOR EACH NODE!
volatile byte state = LOW;

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // My node ID
#define TONODEID      1   // Destination node ID

#define FREQUENCY     RF69_915MHZ

// AES encryption (or not):

#define ENCRYPT       true // Set to "true" to use encryption
#define ENCRYPTKEY    "passwordpassword" // Use the same 16-byte key on all nodes

// Use ACKnowledge when sending messages (or not):
#define USEACK        true // Request ACKs or not

#define CSPIN         10//17
#define IRQPIN        2//20

// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,&SPI);

void setup() {
/*  SPI.setRX(16);
  SPI.setTX(19);
  SPI.setCS(17);
  SPI.setSCK(18);  
  SPI.begin(); */

  Serial.begin(115200);
  while (!Serial)delay(10);
  delay(1000);
  pinMode(IRQPIN, INPUT);
   // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.print("Node ");
  Serial.print(MYNODEID,DEC);
  Serial.println(" ready");
  // Initialize the RFM69HCW:
  if (radio.initialize(FREQUENCY, MYNODEID, NETWORKID)) Serial.println ("Init OK");
  else Serial.println ("Init BAD");
  radio.setHighPower(); // Always use this for RFM69HCW

  // Turn on encryption if desired:

  if (ENCRYPT) {
    radio.encrypt(ENCRYPTKEY);
  }

}

void loop() {
  Serial.println(radio.canSend());
  Serial.print("RSSI: ");
  Serial.println(radio.readRSSI());
  Serial.print("Temp: ");
  Serial.println(radio.readTemperature());
  digitalWrite(LED_BUILTIN, state);
  blink();
  delay(500);
}

void blink() {
  state = !state;
}
```
Result
```
Node 2 ready
Init OK
0
RSSI: -128
Temp: 22
0
RSSI: -128
Temp: 22
0
RSSI: -128
Temp: 22
0
RSSI: -128
Temp: 22
0
RSSI: -128
Temp: 22
```

charles-d-burton, 2022-03-13:
It's a raspberry pi pico so there is not default SPI selection, I have to set the pins on initialization.  I know it's working from a wiring standpoint because I have it working with CircuitPython.  I really think it's something with the IRQ but I'm on the struggle bus with it.

slavicd, 2022-11-13:
> It's a raspberry pi pico so there is not default SPI selection, I have to set the pins on initialization. I know it's working from a wiring standpoint because I have it working with CircuitPython. I really think it's something with the IRQ but I'm on the struggle bus with it.

Did you ever get it to work? I'm trying to do the same.