LowPowerLab/RFM69 #180: Replicating these in RFM63W modules

apurv1305, 2022-07-09:
Was trying to replicate these same operations with RFM63 module (changes made in accordance with the Datasheets), but unable to read all the registers correctly. From what I have observed, certain registers work perfectly as they should (read write operations), but some registers are not consistent with the values when read in the loop. Some registers in-fact un-writable unless you have a 0 at address 0. 
Completely unable to fund any informatino accept the datasheet on the web.
Wanted help to identify what the problem might be.

CODE:
`#include<SPI.h>

#define nssDataPin  10
#define nssConfigPin  9

int delayMicro = 20;
uint8_t address = 0x02;
uint8_t value = 0x03;

void setup() {
  Serial.begin(9600);
  
  pinMode(nssConfigPin, OUTPUT);
  pinMode(nssDataPin, OUTPUT);

  SPI.begin();
  SPI.setClockDivider(SPI_CLOCK_DIV4);
  SPI.setDataMode(SPI_MODE0);
  SPI.setBitOrder(MSBFIRST);
  
  digitalWrite(nssConfigPin, HIGH);
  digitalWrite(nssDataPin, HIGH);  

  writeRegister(address, value);
  delay(1);
}

void loop() {
  // put your main code here, to run repeatedly:
  uint8_t vals = 0;
  
  vals = readRegister(address);
  Serial.println(vals);
  delay(1);
}

void writeRegister(uint8_t address, uint8_t value){
  digitalWrite(nssConfigPin, LOW);
  delayMicroseconds(delayMicro);
  
  uint8_t addr = address << 1;
  SPI.transfer(addr | 0x00);
  SPI.transfer(value);
  
  delayMicroseconds(delayMicro);
  digitalWrite(nssConfigPin, HIGH);
  delayMicroseconds(delayMicro);
}

uint8_t readRegister(uint8_t address){
  uint8_t valReg =0;
  digitalWrite(nssConfigPin, LOW);
  delayMicroseconds(delayMicro);
  
  uint8_t addr = address << 1;
  SPI.transfer(addr | 0x40);
  valReg = SPI.transfer(0);
  
  delayMicroseconds(delayMicro);
  digitalWrite(nssConfigPin, HIGH);
  delayMicroseconds(delayMicro);
  return valReg;
  
}`

OUTPUT:
[image]

apurv1305, 2022-07-11:
could you give some advice in this regard, of any sort?

LowPowerLab, 2022-07-12:
The datasheet is probably the best place to consult when creating a library.