LowPowerLab/RFM69 #184: Receiver bandwidth

PE5PVB, 2022-11-04:
Hi,

Is it possible to set the receiver bandwidth?

LowPowerLab, 2022-11-07:
You can set the bitrate, see library examples of how you might do that.

closed 2022-11-07 by LowPowerLab

PE5PVB, 2022-11-09:
Hi,

Didn't find it. I could only find an option called set300KBPS.
I would like to set the radio to the lowest bandwidth as possible, I use it as a mini spectrum analyser. Maybe you can help me out with the exact commands I have to send by SPI.

PE5PVB, 2023-01-12:
Can you help me by any chance?

LowPowerLab, 2023-01-12:
There's some threads in the forum about 1200 baud setting:
https://lowpowerlab.com/forum/search2/?params=eJwtzcsKwyAUhOFXKd10MwtP7n0a0eOBtNhYjEkp-PDRkN3wwfAbt5uFxeVHVvmebayL0KBHhxYDRkx4ghSIQC2oA_WgATSCpvJY5_DTHD5fL0nKt9Jm38JJh8X_LwkxlRXFy5m7SLtXLOxk5SpiIs813yh1s2ZzB0TTNNw.