LowPowerLab/RFM69 #187: RFM69::initialize returning "false" TRUE

kythre, 2022-12-26:
RFM69::initialize is returning true on a completely bare Arduino Nano. Tried it on two separate Nanos with generic code as below.

`#include <RFM69.h>    //get it here: https://www.github.com/lowpowerlab/rfm69
#include <RFM69_ATC.h>//get it here: https://github.com/lowpowerlab/RFM69
#include <SPI.h>      //included with Arduino IDE (www.arduino.cc)
#include <SPIFlash.h> //get it here: https://www.github.com/lowpowerlab/spiflash

RFM69_ATC radio;

void setup() {
  Serial.begin(115200);
  Serial.println("START!");

  Serial.println("Initializing radio...");

  if (!radio.initialize(RF69_915MHZ,123,200))
    Serial.println("radio.init() FAIL");
  else
    Serial.println("radio.init() SUCCESS");

  radio.enableAutoPower(-80);
}

byte led=0;
void loop() {
  Serial.println("Loop");
  delay(1000);
}`

LowPowerLab, 2022-12-27:
@kythre Thanks, it seems to be intermittent (the SUCCESS vs FAIL with no module). There is no definite check if the module responds the way it should, so I added the missing checks and it should now be resolved.

closed 2022-12-27 by LowPowerLab

kythre, 2022-12-28:
awesome, thank you for the quick update. 
from some quick googling, it looks like those checks are just to check that SPI (and the requisite wiring) is working.
is that correct?

LowPowerLab, 2022-12-28:
The checks are to ensure that what is written to a certain set of registers is the same as what can be read back from those same registers.

commit 06a4a7ba (2022-12-27): Update RFM69.cpp

Fixes #187