LowPowerLab/RFM69 #188: radio.DATALEN always 0

jLynx, 2023-03-27:
Hi, I am having an issue where I am already getting 0 returned from radio.DATALEN

This is from and ESP32 to an ATMEGA328P.
The code on the receiving end (ATMEGA) is:
```
        Serial.println((char *)radio.DATA);
        Serial.println(radio.PAYLOADLEN);
        Serial.println(radio.DATALEN);
        Serial.println(sizeof(radio.DATA));
```
Which prints 
```
123456789abcdefg,1,2,3
0
0
62
```

The code from the sending ESP32 side is:
```
sprintf(buff, "123456789abcdefg,1,2,3");
radio.sendWithRetry(TONODEID, buff, 22);
```

Do you have any ideas on what I am doing wrong here?

LowPowerLab, 2023-03-27:
See [examples](https://github.com/LowPowerLab/RFM69/blob/master/Examples) how DATALEN is used. Also check the [sprintf ](https://cplusplus.com/reference/cstdio/sprintf/)specification for use of arguments.

closed 2023-03-27 by LowPowerLab

jLynx, 2023-03-27:
> See [examples](https://github.com/LowPowerLab/RFM69/blob/master/Examples) how DATALEN is used. Also check the [sprintf ](https://cplusplus.com/reference/cstdio/sprintf/)specification for use of arguments.

I was just hard coding values in sprintf for ease of use in the example. Don't mean to be rude, but I just pointing to the examples does not answer my question, as that's what I believe have done, yet it returns 0. Can you please re open this issue @LowPowerLab

It would help if you could say where I have gone wrong with my example I posted above. As far as I am aware it should contain the length when it it send and receive it on the other side, so I am very confused on why it's set to 0. For example over here in this example https://github.com/LowPowerLab/RFM69/blob/7db941d3c87a27ac096fa19309619d3a7f45a135/Examples/DoorBellMote/DoorBellMote.ino#L154 you are able to check the length is equal to 4. I don't understand how this use differs from mine.

jLynx, 2023-03-27:
Thanks I'll have another look when I get back. Will post full code examples if I can replicate

LowPowerLab, 2023-03-27:
I will add that as long as the library examples work as intended, I'd rather you start with that and build ontop known working code, rather than show code that apparently doesn't work and spend the time to point out the errors. Again this is not the place to learn c++ in an issue feed, but rather point library problems and fix them if there is indeed a problem.

jLynx, 2023-03-27:
I'm talking about the examples themselves

jLynx, 2023-03-28:
So for anyone else who runs into this issue, the problem was that I was calling ``radio.DATALEN`` after I had called ``radio.sendACK()`` which caused the length to be set back to 0.

---

But back onto the broken examples @LowPowerLab .
The struct example seems to be broken for me:
https://github.com/LowPowerLab/RFM69/blob/7db941d3c87a27ac096fa19309619d3a7f45a135/Examples/Struct_send/Struct_send.ino#L166
https://github.com/LowPowerLab/RFM69/blob/7db941d3c87a27ac096fa19309619d3a7f45a135/Examples/Struct_receive/Struct_receive.ino#L138

I get the error: `Invalid payload received, not matching Payload struct`

When I debug it further I see `Sending struct (12 bytes)` from the sender side and then on the receiver side `radio.DATALEN` is 12 bytes, but `sizeof(Payload)` is only 10 bytes, hence why I am seeing that message.

So what's odd is it seems to be sending the correct information since it sent 12 bytes and received 12 bytes, but when it trys to see the size of the struct type its only 10 bytes.

If I disable that check and force it to cast to ``Payload`` I get the following response 
```nodeId=1 uptime=3960143872 temp=ovf``` when it should be ```nodeId=1 uptime=3960143872 temp=91.23```

Let me know if you need any more information.

_Note: I have changed the NODEID, NETWORKID and GATEWAYID_