LowPowerLab/RFM69 #189: RFM69 on STM32G0

cece99, 2023-07-31:
I'm trying to use an RFM69 module with a STM32G030F6P6 mcu, but it looks like there is some kind of issue when I'm trying to send a packet.

This is how I'm setting up the Radio (PA4 is the spi chip select and PC15 is DI0):
```
#include <RFM69.h> 
#include <RFM69_ATC.h> 

#define NODEID      99
#define NETWORKID   100
#define GATEWAYID   1

#define FREQUENCY     RF69_868MHZ

#define ENCRYPTKEY    "sampleEncryptKey"
#define IS_RFM69HW_HCW  

#define SERIAL_BAUD 115200

RFM69 radio (PA4, PC15);

void setup() {
  pinMode(PA0, OUTPUT);
  pinMode(PA13, OUTPUT);
  digitalWrite(PA0, LOW);
  Serial.begin(SERIAL_BAUD);
  radio.initialize(FREQUENCY,NODEID,NETWORKID);
  radio.setHighPower();
  radio.encrypt(ENCRYPTKEY);
}
```

These are the tests I've done:

If I put in the void loop this code:
```
void loop() {
    radio.writeReg(0x58, 0x1B);
}
```

I can see using a logic analyzer connected to the SPI bus that some packets are in fact exchanged:
[image]

I've also tried reading the register with:
```
void loop() {
    radio.readReg(0x58);
}
```

Which gets me on the logic analyzer this:
[image]

Now if I try to send some data with:
```
void loop() {
    radio.sendWithRetry(1, "Hi", 1);
}
```

The bus goes absolutely crazy and no data is transmitted:
[image]

This is the packet that is being send over and over again in the SPI bus:
[image]

It looks like reading and writing registers works correctly, but sending a frame does not.

cece99, 2023-07-31:
If I try running:
```
    if (radio.readReg(0x58) == 0x1b)
        digitalWrite(PA13, HIGH);
```
Where PA13 is the onboard LED, the LED does not turn on.
I believe there is some kind of issue in comparing the readReg to 0x1b, because I'm confirming with the logic analyzer that the mcu is receiving 0x1b via SPI.

Regarding 
```
void loop() {
    radio.sendWithRetry(1, "Hi", 1);
}
```
I think that one of the problems is in RF69.cpp in line 413:
`while ((readReg( ) & RF_IRQFLAGS2_PACKETSENT) == 0x00);`
The condition for some reason is always false and the while loop goes forever.

LowPowerLab, 2023-08-01:
Please start from known working examples in the library, there is no known bugs with sending simple packets like that.

closed 2023-08-01 by LowPowerLab