LowPowerLab/RFM69 #193: Errors in the library

SaveliiLisnichenko, 2024-10-23:
Hello,
Yesterday I tried to use your library and when I try to upload the code it gives me the following output:
```
In file included from /home/savalio/Arduino/libraries/RadioHead/RH_RF69.h:12,
                 from /tmp/.arduinoIDE-unsaved2024923-6804-1v3hwm6.d3h3k/rf69_client/rf69_client.ino:15:
/home/savalio/Arduino/libraries/RadioHead/RHGenericSPI.h:118:34: error: macro "attachInterrupt" requires 3 arguments, but only 1 given
  118 |     virtual void attachInterrupt() {};
      |                                  ^
In file included from /home/savalio/.arduino15/packages/esp32/hardware/esp32/3.0.5/cores/esp32/Arduino.h:252,
                 from /tmp/arduino/sketches/74AAA254F45FD4D4C35AFB27A33C9AEF/sketch/rf69_client.ino.cpp:1:
/home/savalio/.arduino15/packages/esp32/hardware/esp32/3.0.5/cores/esp32/io_pin_remap.h:44: note: macro "attachInterrupt" defined here
   44 | #define attachInterrupt(pin, fcn, mode)         attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
      | 
In file included from /home/savalio/Arduino/libraries/RadioHead/RHSPIDriver.h:10,
                 from /home/savalio/Arduino/libraries/RadioHead/RH_RF69.h:13:
/home/savalio/Arduino/libraries/RadioHead/RHHardwareSPI.h:66:26: error: macro "attachInterrupt" requires 3 arguments, but only 1 given
   66 |     void attachInterrupt();
      |                          ^
/home/savalio/.arduino15/packages/esp32/hardware/esp32/3.0.5/cores/esp32/io_pin_remap.h:44: note: macro "attachInterrupt" defined here
   44 | #define attachInterrupt(pin, fcn, mode)         attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
      | 
exit status 1

Compilation error: exit status 1
```

Here's the example code I was using (RadioHead Library -> rf69_client.h example):

```
// rf69_client.pde
// -*- mode: C++ -*-
// Example sketch showing how to create a simple messageing client
// with the RH_RF69 class. RH_RF69 class does not provide for addressing or
// reliability, so you should only use RH_RF69  if you do not need the higher
// level messaging abilities.
// It is designed to work with the other example rf69_server.
// Demonstrates the use of AES encryption, setting the frequency and modem 
// configuration
// Tested on Moteino with RFM69 http://lowpowerlab.com/moteino/
// Tested on miniWireless with RFM69 www.anarduino.com/miniwireless
// Tested on Teensy 3.1 with RF69 on PJRC breakout board

#include <SPI.h>
#include <RH_RF69.h>

// Singleton instance of the radio driver
RH_RF69 rf69;
//RH_RF69 rf69(15, 16); // For RF69 on PJRC breakout board with Teensy 3.1
//RH_RF69 rf69(4, 2); // For MoteinoMEGA https://lowpowerlab.com/shop/moteinomega
//RH_RF69 rf69(8, 7); // Adafruit Feather 32u4

void setup() 
{
  Serial.begin(9600);
  while (!Serial) 
    ;
  if (!rf69.init())
    Serial.println("init failed");
  // Defaults after init are 434.0MHz, modulation GFSK_Rb250Fd250, +13dbM (for low power module)
  // No encryption
  if (!rf69.setFrequency(433.0))
    Serial.println("setFrequency failed");

  // If you are using a high power RF69 eg RFM69HW, you *must* set a Tx power with the
  // ishighpowermodule flag set like this:
  //rf69.setTxPower(14, true);

  // The encryption key has to be the same as the one in the server
  uint8_t key[] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
                    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08};
  rf69.setEncryptionKey(key);
}

void loop()
{
  Serial.println("Sending to rf69_server");
  // Send a message to rf69_server
  uint8_t data[] = "Hello World!";
  rf69.send(data, sizeof(data));
  
  rf69.waitPacketSent();
  // Now wait for a reply
  uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];
  uint8_t len = sizeof(buf);

  if (rf69.waitAvailableTimeout(500))
  { 
    // Should be a reply message for us now   
    if (rf69.recv(buf, &len))
    {
      Serial.print("got reply: ");
      Serial.println((char*)buf);
    }
    else
    {
      Serial.println("recv failed");
    }
  }
  else
  {
    Serial.println("No reply, is rf69_server running?");
  }
  delay(400);
}
```

SaveliiLisnichenko, 2024-10-23:
Update: Just tried adding the pin definitions to the initialization of the RF69 module, it sill

LowPowerLab, 2024-10-24:
Sorry but that library is RadioHead, not LowPowerLab_RFM69.

closed 2024-10-24 by LowPowerLab

SaveliiLisnichenko, 2024-10-24:
Your library is the part that's actually generating the error, I only used the example from RadioHead, and the example doesn't use the RadioHead library in any way, shape or form

LowPowerLab, 2024-10-24:
Really? Where and how?

SaveliiLisnichenko, 2024-10-24:
Wait no sorry I thought the error said that it was your library