LowPowerLab/RFM69 #196: ESP32-C3 - RISC-V RISING edge interrupt is not cleared in isr0 handler

simonantonio, 2025-01-11:
I've been trying to get the library working with some ESP32-C3's however it would stack-dump every time the RFM69 raised an interrupt.

Turns out this chip does not clear the interrupt so it kept triggering over and over leading to WDT resets. 

my hacky fix (because my c++ foo is weak - and I couldn't work out how to clear the flag in `GPIO.status_w1tc`

```
ISR_PREFIX void RFM69::isr0() {
  int interruptPin = _instance->_interruptPin;
  detachInterrupt(interruptPin); // disables the interrupt until we are done
                                 // and clears the flag

  _haveData = true;
  if (_instance->_isrCallback)
    _instance->_isrCallback();

  attachInterrupt(interruptPin, DavisRFM69::isr0, RISING); // re-enable the interrupt
}
```

---
Hopefully this is helpful - I've been using this with a ESP32-C3 SuperMini paired to an RFM69HCW

[image]

| ESP32-C3 PIN | RFM69 PIN |
|-----------|-----------|
| 5 (MISO)  | MISO      |
| 6 (MOSI)  | MOSI      |
| 7 (SS)    | NSS       |
| 4 (SCK)   | SCK       |
| 3         | DIO1      |

simonantonio, 2025-02-07:
> The isr0 is setup on RISING edge, why would the ESP trigger the isr0 continuously? `attachInterrupt(_interruptNum, RFM69::isr0, RISING);`

I'm unsure, it doesn't clear the interrupt is my understanding - something specific to these esp32-c3 modules
I scoped the interrupt pin its only triggered once, however the flag doesn't get cleared when the ISR runs, so it reruns until the WDT resets the esp. 

---

Regardless I'll close this since its not a problem with your library but upstream. Maybe this will help someone if they come unstuck like I did.

closed 2025-02-07 by simonantonio

LowPowerLab, 2025-02-07:
Sounds good, thanks. It will be here for reference in case someone else needs it.

simonantonio, 2025-09-22:
> Also using an esp32-c3 here.
> 
> [@simonantonio](https://github.com/simonantonio) was your isr0 hack sufficient to work around this problem? Is there a tipo in your code? `DavisRFM69` in the last line does not look right. Is it rather `attachInterrupt(interruptPin, RFM69::isr0, RISING);` instead ??

Sure that should work, my original snippet was from a project using a fork of this lib.